openradar/xradar #1: Next initialization steps

kmuehlbauer, 2022-09-01:
@mgrover1 Thanks for setting this up.

What would be the next steps? Please extend this list.

- [ ] release on PyPI

mgrover1, 2022-09-01:
I put together a 0.0.0 release here - just so we can reserve the namespace

kmuehlbauer, 2022-09-01:
@mgrover1 It works now to create the correct version. But we would need to keep an eye on it.

kmuehlbauer, 2022-09-01:
@magrover1 Could you elaborate how the bridge to RTD is closed. We might need to tweak this a bit to get the documentation build from the latest commit/tag instead of some source without the version.py as of now.

kmuehlbauer, 2022-09-13:
@mgrover I think we've passed almost all initalization steps. How should we handle the license entity in "LICENSE"? Should we move to "openradar developers"?

kmuehlbauer, 2022-09-13:
Örks, two times misspelled @mgrover1 in a row. :face_with_spiral_eyes:

mgrover1, 2022-09-13:
> @mgrover1 I think we've passed almost all initalization steps. How should we handle the license entity in "LICENSE"? Should we move to "openradar developers"?

That sounds good!

kmuehlbauer, 2022-09-14:
Done! :+1:

closed 2022-09-14 by kmuehlbauer