openradar/xradar #17: Improve Documentation

kmuehlbauer, 2022-09-14:
We now have a minimal documentation, yikes https://docs.openradarscience.org/projects/xradar/
It already looks very nice, but can be improved. 

What's missing list (please extend):

- [ ] direct links to the current standards
- [ ] description of the data model (mandatory, optional)
- [ ] fleshed out description how the importers are working
- [ ] showcase of implementation of an example accessor
- [ ] more notebook tutorials (what?)
- [ ] tbc

mgrover1, 2022-09-14:
Thanks for opening this - I agree here. We can definitely start with these points and continue to expand.

mgrover1, 2022-09-22:
@kmuehlbauer - here is the repo setup to host sample datasets https://github.com/openradar/open-radar-data . I think we can use this in our examples and such, across these different projects.

mgrover1, 2022-09-29:
@kmuehlbauer - when I was working on an RHI example, I noticed how helpful it would be if we had azimuth with range as the dimensions instead of time... so one could use the following

```python
radar.sel(azimuth=270, method='nearest')
```

Is this something we can consider? Or does this violate the data model?

kmuehlbauer, 2022-09-29:
@mgrover1 

Yes, that's what we did in wradlib already. Internally we can do anything we want if only we write out in the correct manner.

Try to open with keyword argument `first_dim="auto"`. This will get either "azimuth" or "elevation" as first dim.

kmuehlbauer, 2025-06-24:
@egouden We can always improve 😄. But yes, maybe it's better to open dedicated issues nowadays and close this one.

closed 2025-06-24 by kmuehlbauer