openradar/xradar #25: Georeferencing Functionality

mgrover1, 2022-09-22:
* xradar version:
* Python version:
* Operating System:

### Description

For plotting purposes, it would be helpful to have the x, y, z as well as lat, lon, altitude variables for a given dataset. These are calculations that have traditionally lived in wradlib (requiring it after loading in the dataset), and Py-ART (lazily computing directly in the radar object).

Should this remain Py-ART/wradlib, or is this something we can implement here? From a plotting perspective, these fields would be very convenient to have. Even if it was accessible via some sort of `.postprocess()` routine... 

@kmuehlbauer - eager to hear your thoughts here.

kmuehlbauer, 2022-09-22:
That is something I was already thinking about when working on the cfradial1/odim example notebooks.

Georeferencing is a crucial part of infrastructure and from my perspective it should live within xradar. Even if that might open a can of worms here. 

My suggestion would be to just provide the aeqd projection (which does not involve proj or gdal) within xradar and leave the heavy lifting to other packages (wradlib/gdal or Py-ART/proj or cartopy).

mgrover1, 2022-09-22:
I agree here - I can draft up a PR here. Should we have these be directly integrated into the xradar object, as with the Py-ART radar object, where it will be lazy-loaded? Or should we have a method that will explicitly apply this?

kmuehlbauer, 2022-09-22:
Good question Let's work on the implementation first, we can decide later how to bind that functionality to the object.

We would need to carefully design this with respect to effective earth radius, k-factor etc etc

kmuehlbauer, 2022-09-28:
First step taken with #28 with the work by Max. https://docs.openradarscience.org/projects/xradar/en/latest/notebooks/plot-ppi.html :tada:

closed 2023-03-01 by mgrover1