openradar/xradar #27: Implement remaining importers from wradlib

kmuehlbauer, 2022-09-26:
- rainbow
- sigmet
- gamic hdf5
- furuno scn/scnx

mgrover1, 2022-09-26:
Can we make this more general? I would like to add the NEXRAD one here too, as well as other readers from Py-ART

kmuehlbauer, 2022-09-26:
@mgrover1 Sure, please add to the first post.

jthielen, 2022-09-27:
@mgrover1 Were you planning on implementing the NEXRAD level 2 importer from Py-ART in the near future? I am hoping to adopt the xradar data model in a downstream package ([OpenMosaic](https://github.com/jthielen/OpenMosaic)), but would be blocked for now by not having that particular reader. If you wouldn't otherwise be already be working on it, I could work towards putting together a PR.

mgrover1, 2022-09-27:
@jthielen - if you are interested, go for it! My next task is the georeferencing, so if you could tackle the level2 reader first, that would be fantastic (and very much welcome :) ). Feel free to use the existing functions/such in Py-ART.

kmuehlbauer, 2022-10-21:
Yes, maybe it is more visible then.