openradar/xradar #31: Dataset/DatArray validation

kmuehlbauer, 2022-09-27:
We might think about applying [xarray-schema](https://github.com/xarray-contrib/xarray-schema) for validation with the standard.

kmuehlbauer, 2022-09-30:
@mgrover1 brought this up in https://github.com/openradar/xradar/issues/17#issuecomment-1262483522 how we should handle the time vs. range and azimuth/elevation vs range. We can do fancy things with the latter.

From my (wradlib based) perspective we can do anything with the xradar model as long as we write standard conforming files. The current reader implementation allows with kwarg `first_dim="auto"` to automatically switch to the azimuth/elevation vs range layout.

For writing we should facilitate xarray-schema before using a dedicated `to_cfradial2` or `to_odim` writer function.

mgrover1, 2022-09-30:
Agreed here!