openradar/xradar #38: Georeferencing follow-up issue

kmuehlbauer, 2022-09-30:
to not forget the items from #25: 

- [ ] Move cartesian coordinate definition into model see https://github.com/openradar/xradar/pull/28#discussion_r981978559
- [ ] use model functionality to create Dataset
- [ ] docstring fixes
- [ ] tbc

mgrover1, 2022-10-12:
For georeferencing... @kmuehlbauer what would you think if we had an optional cartopy dependency? To deal with these projections?

kmuehlbauer, 2022-10-12:
@mgrover1 Do you refer to map-making? Could you elaborate a bit more with regard to what functionality you have in mind?

mgrover1, 2022-10-12:
I am thinking in terms of projecting data as well - see the MetPy functionality here https://unidata.github.io/MetPy/latest/examples/XArray_Projections.html

kmuehlbauer, 2022-10-12:
Yes, I see, they use an accessor to get a cartopy crs object.

From my perspective we would just need to add the aeqd cartopy crs to Dataset/DataArray when creating xyz-coordinates if cartopy is available.

mgrover1, 2022-10-12:
Yup! That makes sense!

mgrover1, 2022-10-21:
Sorry for the delay here @kmuehlbauer ... prepping for our ARM/ASR meeting next week.

kmuehlbauer, 2022-10-21:
No worries @mgrover1. It's great we are this far already.

kmuehlbauer, 2023-02-14:
> From my perspective we would just need to add the aeqd cartopy crs to Dataset/DataArray when creating xyz-coordinates if cartopy is available.

@mgrover1 I've though about that the last days and tinkered a bit with `pyproj`. I've ended up with a workflow where we can use `pyproj` to:

- georeference the data using the derived latitude-dependend earth-radius
- write_crs (as `spatial_ref` - Variable)
- get_crs (retrieve `spatial_ref` or create on the fly if not available)

The would be able to wrap that into a `cartopy.crs.Projection` and use it for plotting with cartopy.

That would leave us with only `pyproj`/`proj` as additional dependencies. We could even make this optional. I'll see if I can workout a PR from that.

kmuehlbauer, 2023-02-15:
@mgrover1 I've setup a PR with a minimal pyproj-based solution in #87.

kmuehlbauer, 2023-03-01:
@mgrover1 We might close this issue and #25, as we have the basic workable georeferencing available now. WDYT?

mgrover1, 2023-03-01:
Agreed! Let's do it.

closed 2023-03-01 by mgrover1