openradar/xradar #40: Add a NEXRAD reader

mgrover1, 2022-10-21:
### Description

Add a backend for NEXRAD level 2 and level 3 files

### What We Want to Do

NEXRAD data is heavily used within the radar community, and it would be helpful to have a reader for this data in xradar.

Py-ART has readers for this, including the
- [NEXRAD Level 2 Reader](https://github.com/ARM-DOE/pyart/blob/main/pyart/io/nexrad_archive.py)
- [NEXRAD Level 3 Reader](https://github.com/ARM-DOE/pyart/blob/main/pyart/io/nexrad_level3.py)

@jthielen mentioned he would be interested in helping with this (see #27)... others can help too!

jthielen, 2022-10-21:
> @jthielen mentioned they would be interested in helping with this (see #27)... others can help too!

Yes indeed! I still hope to get around to implementing the Level 2 reader, but if someone beats me to it, that'd work too! Also, I wouldn't have need for the Level 3 reader in my own work, so I will refrain from taking on that one.

mgrover1, 2023-02-27:
@jthielen any updates here? If not, I can go ahead and start drafting something

jthielen, 2023-02-27:
> @jthielen any updates here? If not, I can go ahead and start drafting something

Thanks for the ping! My work on this has unfortunately stalled recently due a few other things taking higher priority, but I can least share what I've got so far: https://github.com/jthielen/xradar/tree/nexrad-level2. Currently, a majority (but certainly not all) of the file reader is implemented (what remains has `TODO` comments), whereas the backend entrypoint isn't (but should be much easier than the file reader to code). Feel free to take what I have as a starting point to build off of if you'd want this to progress sooner rather than later, otherwise, I believe I should have the chance to return to this in mid-March.