openradar/xradar #41: Add NIMROD support

alxmrs, 2022-10-22:
We'd like to be able to read radar data from the U.K. Met office (via the CEDA archive) in the NIMROD format. CEDA provides some software to do this now, which seems fairly unsupported: https://data.ceda.ac.uk/badc/ukmo-nimrod/software/python. There are other projects that parse NIMROD data, however, they typically use a copy-left license, which is a non-starter for many commercial applications. Last, there doesn't seem to be a library that directly reads NIMROD data into XArray – a perfect opportunity for xradar. 

@mgrover1 -- correct me if I'm wrong about any of this background :) 

If it makes sense, @mahrsee1997 and I would be interested in contributing a Nimrod parser to this library.

mgrover1, 2022-10-22:
I think this sounds great - let me know if you have any questions. Best place to start would be looking at our data model docs https://docs.openradarscience.org/projects/xradar/en/latest/datamodel.html and looking at the backends already implemented! Happy to help with this process 😁

kmuehlbauer, 2022-10-22:
IIUC that's a gridded format, which doesn't fit into our radial model.

We would need to discuss how we can add this as an additional gridded model.

@DanielMichelson mentioned once that WMO is also working on extending the forecoming standard in that sense.

kmuehlbauer, 2022-10-22:
After reading a bit more on NIMROD it seems that also single station polar data is available. That would definitely fit here directly into the model. @alxmrs Is this correct, that you are referring do that polar data?

kmuehlbauer, 2022-10-22:
Just for reference, here is the link to sample files https://data.ceda.ac.uk/badc/ukmo-nimrod/hdf5-samples/met-office-odims-sample-files

Those are announced as ODIM_H5 format.

egouden, 2023-04-20:
I cannot register to check the data. But it seems to be ODIM data. There is already a reader for ODIM radial data in xradar.

For 2D ODIM products, you can use [this](https://corteva.github.io/rioxarray/stable/).

There might be a UK radial format, but this would be converted to ODIM/FM301 by them.