openradar/xradar #48: Site-coordinates at Sweep-Level

kmuehlbauer, 2022-11-15:
In CfRadial2-standard the sweep-groups do not contain the site-coordinates (`latitude`/`longitude`/`altitude`). Although that makes perfectly sense when reading the whole DataTree, it might be not as perfect for the single-sweep reading via `xr.open_dataset`. The site-coordinates (at least `latitude` and `altitude`) might be used for georeferencing.

I'd suggest to enable the xarray-backends to add the site-coordinates to the returned Dataset. We have at least the following options:

- return site-coords per default in any case
- return site-coords per kwarg-request per default
- return site-coords per kwarg-request not per default
- ?

Thoughts?

mgrover1, 2022-11-15:
I like the idea of keeping that at the root level... we can add an attribute perhaps? But keeping at the root dataset level makes sense

kmuehlbauer, 2022-11-15:
@mgrover1 Yes, they should definitely stay at the root group. And they should not show up in sweep groups inside DataTree.

But it would be great if they could show up when single sweeps are read via `xr.open_dataset`.

I'll create a PR with a proposal.

PR #49 (2022-11-21): CfRadial1 reader kwargs refactor, add to_cfradial2 exporter 
1. CfRadial1 reader kwarg refactor:

```python
"""
    first_dim : str
        Can be ``time`` or ``auto`` first dimension. If set to ``auto``,
        first dimension will be either ``azimuth`` or ``elevation`` depending on
        type of sweep. Defaults to ``auto``.
    optional : bool
        Import optional mandatory data and metadata, defaults to ``True``.
    site_coords : bool
        Attach radar site-coordinates to Dataset, defaults to ``True``.
"""
```

```python
swp = xr.open_dataset(fname, group="sweep_0", site_coords=Fa [... cut]
closed 2022-11-21 by kmuehlbauer