openradar/xradar #55: Angle Reindexing

kmuehlbauer, 2022-11-23:
* xradar version: 0.0.8

### Description

As a legacy from wradlib we have complex code for angle reindexing in xradar's codebase.

#### High precision angle coordinates

As radar angle coordinates (`azimuth` or `elevation`) are measured constantly by different techniques of detection of the antenna pointing direction the values in the data files are mostly floating point numbers. In many cases these floating point numbers are not rounded to a certain decimal but keep the full possible range of the used dtype.

Problems of that:

- 1D angle coordinate arrays yield no equidistant vector.
- 1D angle coordinate arrays are not equivalent for different timesteps but same scan setup

#### Missing rays, duplicate or additional rays

Sometimes rays (even sectors) are missing from the dataset, sometimes there are duplicate rays. Another problem with radar data are additional rays, which I call "antenna hickup" (two rays measured with within one resolution interval).

#### What is angle reindexing?

Angle reindexing takes care of these problems by trying to determine the wanted layout from the radar metadata and the angle coordinates. With that newly created angle coordinate xarray machinery is used to reindex the radar moment data to that by nearest neighbor lookup (up to a tolerance). Missing rays will be filled with NaN.

#### Why should it be used?

For most operations this is not a real problem. It will turn into a problem, if you want to stack your xarray.Dataset radar data on a third dimension (eg. `time`, by using `open_mfdataset`). Then all coordinates need alignment to keep things simple and manageable (eg. `azimuth=[0.5, 1.5, 2.5,..., 359.5]`)

#### How should we treat it?

Currently the reindexing code relies on some internals which make things a bit hard to maintain. My suggestion would be to disentangle the reindexing code from the internals but feed the needed values as parameters. Then every reader can call this per activated `reindex_angle` kwarg.

### Discussion

I'm very open for discussion to find the most optimal way to provide that feature.

mgrover1, 2022-11-23:
@kmuehlbauer I would agree that going with the more simple option here makes sense! For now,  I have been leaning toward "dropping duplicates" but this is not a robust way of going about it...

kmuehlbauer, 2022-11-24:
@mgrover1 With #53, #56, #57 and #58 all readers are aligned to the same convention (parameters etc).

I'll strip out the reindexing algorithm and make this work as a drop in for the different readers.

kmuehlbauer, 2022-12-08:
@mgrover1 I've created some utility functions in the context of angle reindexing. Shall I put them into `util` first until we decide for the final destination?

mgrover1, 2022-12-08:
Util sounds like the right place for now!

kmuehlbauer, 2022-12-11:
With #62 in we have angle reindexing (with time inter-/extrapolation) available. We might need to update/show example use cases for the different readers.

PR #62 (2022-12-11): ENH: Add angle reindexing util functions, add example notebook
Closes #55 

Todo:

- [x] add tests
- [x] replace old reindexing code with new functionality
closed 2022-12-11 by kmuehlbauer