openradar/xradar #76: Actual azimutal start/end angles lost when importing Rainbow data

egouden, 2023-02-06:
When importing Rainbow data, for a given ray, the information on actual azimuth start and azimuth end is lost because the actual angle step is slightly different than 1.

kmuehlbauer, 2023-02-06:
@egouden Thanks for raising. I've a couple of questions.

- How did you read the data?
- Are you referring to `startangle` and `startangle` from `rayinfo`-fields?
- For what purpose do you need these arrays?

For the CfRadial2 standard (which we try to aim for with the data model) there is only one azimuth array which is given as beam-center. So we convert those two arrays to beam-center values.

egouden, 2023-02-06:
What if one wants to write to an ODIM format with start and end azimuth angles?

kmuehlbauer, 2023-02-06:
Currently ODIM_H5 v2.2 is supported, which writes time, azimuth and elevation arrays for start/stop ray.

As those are calculated from the center values, they might not be equivalent.

kmuehlbauer, 2023-02-06:
Exact round-tripping isn't guaranteed.

kmuehlbauer, 2023-08-16:
I'm going to close this. We might handle this in the future by keeping azimuth_start and azimuth_stop within the data but need to check first what implications this has. Until then we use the momentary solution.