openradar/xradar #88: Possibility of xmovie util

zssherman, 2023-02-15:
Hello!

So we were discussing for ACT that a movie creator for radar data/images would be useful, but wouldn't really fall into the scope of ACT. With xradar utilizing xarray, which is what xmovie does:
https://xmovie.readthedocs.io/en/latest/examples/quickstart.html

Would this be something we'd want in xradar? Or and example of?

kmuehlbauer, 2023-02-15:
An example notebook, how to utilize xmovie with xradar would be really great.

mgrover1, 2023-02-15:
Yeah, I agree, having an example showing how to use this with xradar would be great!

zssherman, 2023-02-15:
The notebooks in xradar run during testing correct? Trying to think how to have xmovie, I assume optional dependency, or something but wasn't sure how to handle it for the notebook example

kmuehlbauer, 2023-02-15:
Yes, they are running during testing. Need to add xmovie to ci/unittests.yml.

They are also run on readthedocs.  Need to add xmovie to environment.yml.

zssherman, 2023-02-15:
Gotcha sounds good!

zssherman, 2023-02-15:
Made the mistake, xmovie hasn't been updated yet for datatrees, so closing for now.

zssherman, 2023-02-15:
After discussion with @mgrover1, will leave open, until Max merges a multi sweep example.

kmuehlbauer, 2023-02-16:
Yes, the xmovie guide might just be appended to that notebook. :+1:

kmuehlbauer, 2025-07-11:
There is basic matplotlib animation in https://docs.openradarscience.org/projects/xradar/en/stable/notebooks/multiple-sweeps-into-volume-scan.html.

xmovie seems unmaintained https://github.com/jbusecke/xmovie/issues/163. I've had nice experience with https://github.com/ahuang11/streamjoy lately.

Closing this as resolved.