openradar/xradar #98: Version Handling in `to_odim`

kmuehlbauer, 2023-03-08:
* xradar version: 0.1.0
* Python version: 3.11 
* Operating System: Linux

### Description

The output `to_odim` is currently assuming ODIM_H5 V2.2. We should discuss how we want to enable output of V2.3 and V2.4 and possible further versions.

One problem with that was raised by @egouden in https://github.com/openradar/xradar/pull/97#issuecomment-1460451436.

egouden, 2023-03-08:
The ODIM writer is now fully compatible with ODIM 2.2 (2014), which is good !

The main interest of ODIM 2.4 (2021) is the definition of some mandatory metadata, useful for data assimilation in NWP.

We could keep only 2.4, because it is cheaper and the format is backward compatible.

We could also release quickly 2.4, by focusing on the mandatory metadata.

Some mandatory metadata in 2.4 might not be available in our model but I do not think this is an issue. The missing information can be provided as parameters, like the source parameter in 2.2.

syedhamidali, 2026-10-03:
Picking this up again, since it's been quiet since 2023. Where we stand on `main`:

**1. Version.** `to_odim` still writes ODIM_H5 2.2 only: `Conventions = "ODIM_H5/V2_2"`, `what/version = "H5rad 2.2"` (the `# todo: parameterize version` is still there, [odim.py#L180-L199](https://github.com/openradar/xradar/blob/21c034c/xradar/io/export/odim.py#L180-L199)), and `rstart` in km ([L247](https://github.com/openradar/xradar/blob/21c034c/xradar/io/export/odim.py#L247)). The reader side already handles km (2.2) vs m (2.4) since #196/#197.

What 2.4 would need on top ([ODIM_H5 v2.4](https://www.eumetnet.eu/wp-content/uploads/2021/07/ODIM_H5_v2.4.pdf)):
- `ODIM_H5/V2_4` / `H5rad 2.4` (Section 4.1, Table 1), `where/rstart` in meters (Table 4)
- `NOD` in `what/source` is mandatory for single-site data (Release notes; Table 1: *"For single-site data the NOD identifier is mandatory … All identifiers assigned to a given radar shall be provided."*)
- the mandatory `how` attributes listed for a polar volume (Section 7.1, Table 19): `antgainH/V`, `beamwH/V`, `frequency`, `NI`, `radconstH/V`, `RXbandwidth`, `RXlossH/V`, `scan_count` at top level, and `scan_index`, `startazA`, `stopazA` per dataset

Some of these we have in the model (`radar_parameters` beam widths/frequency, `nyquist_velocity`, the ray start/stop angles), others (`radconstH/V`, `RXlossH/V`, `RXbandwidth`) usually not, so, as @egouden suggested above, they would come in as parameters.

**2. Source.** Since #96 `source=` is a required string. Meanwhile the ODIM reader parses `/what/source` into root attrs (`WMO` → `wmo__id`, `NOD` → `node`, `ORG` → `wmo__originating_centre`, `WIGOS` → `wmo__wsi`, `PLC` → `site_name`, [odim.py#L214-L220](https://github.com/openradar/xradar/blob/21c034c/xradar/io/backends/odim.py#L214-L220)), but the writer doesn't use them. So an ODIM file read with xradar can't be written back without retyping its own source string. The round trip currently needs a human in the loop 🙂

Proposal, in two steps:
1. Small PR: with `source=None`, `to_odim` builds `what/source` from those root attrs; it only raises if none are there, and an explicit `source` still wins. Plus a read → write → read round-trip test.
2. Version PR: `to_odim(..., version="2.4")` setting the strings, `rstart` units and the mandatory `how` attributes (missing ones as kwargs, or an error listing what's missing).

@kmuehlbauer for step 2: keep 2.2 as default and add 2.4, make 2.4 the default, or go 2.4-only as @egouden proposed? 2.4 adds the metadata NWP users need (per the release notes), but 2.4-only would make `to_odim` fail for trees that lack e.g. `NOD`.

kmuehlbauer, 2026-10-03:
I'd always target the latest version of the standard when writing, so fully agree with @egouden.