openradar/xradar #101: implement ODIM BUFR readers

jfigui, 2023-03-22:
At the last OPERA ET meeting it emerged that there is a real interest for xradar to have and ODIM BUFR reader, in particular to read the French radar data. 
I am fully aware that that is not an easy task but considering that there are plans to open the OPERA data archive and that, particularly at the early stages of the OPERA program, many files are in BUFR format this would be very welcomed by the European radar community.

kmuehlbauer, 2023-03-22:
@jfigui Thanks for bringing this to attention.

wradlib used to read BUFR back in the old days, but dropped support because of all kind of issues. The bufr related python ecosystem evolved much since then, so this might not be an issue any more. 

If you can point to some resources on how to read french bufr data that would be great. Also docs on how to handle tables etc. would be nice to have.

jfigui, 2023-03-22:
@kmuehlbauer ,

you probably know that but there are resources linked to OPERA Bufr here:
https://www.eumetnet.eu/activities/observations-programme/current-activities/opera/

In particular:
https://www.eumetnet.eu/wp-content/uploads/2017/01/OPERA_bufr_description_2012.pdf

https://www.eumetnet.eu/wp-content/uploads/2017/03/bufr_3.2.zip

I can probably send you some examples of polar volume bufr files as well if needed.

kmuehlbauer, 2023-03-22:
Yeah, but good it's now linked here. I've thought there would be some low hanging fruit with new packages like [pdbufr](https://github.com/ecmwf/pdbufr) or even directly with [eccodes](https://github.com/ecmwf/eccodes). The code on that eumetnet-page is rather old, unfortunately.

It would be great if you could provide some bufr polar volumes. Do they need special tables or can they be read with the standard tables?

jfigui, 2023-03-22:
I attach a copy of the tables that are used at Météo-France
[tables_bufr_opera.zip](https://github.com/openradar/xradar/files/11039850/tables_bufr_opera.zip)

jfigui, 2023-03-22:
One thing to keep in mind is that as far as I understood those files are not a single bufr file with multiple fields but a concatenation of bufr files. To be able to decode them properly you have to first cut the file and then read the bufr. 
Another problematic issue is that the field data is compressed using z lib so even when you manage to decode the bufr you still have to decompress the data.

kmuehlbauer, 2023-03-22:
I feel like I'm remembering now, why we've dropped bufr in wradlib :grimacing:

mgrover1, 2023-03-22:
I am wondering if this might be of use here 
https://github.com/pytroll/trollbufr

jfigui, 2023-03-22:
Last time I checked trollbufr was not able to decode the Météo France BUFR. Among other things it was not prepared to read BUFR version 2 or 3 but it can be starting point.

kmuehlbauer, 2023-03-22:
There's also [pybufrkit](https://github.com/ywangd/pybufrkit), which might be worth a test.

rzambranap, 2023-06-30:
Hey everyone, have we had any success in this field ? I recently received a load of bufr radar files from météofrance, and I'm having little to no success in reading them
 with pybufrkit or even the opera bufr linux software listed in one of the comments up there. For me it is quite important and necessary to read these files and I use a python environment, if someone could give me any clues of how to move forward, I'll gladly be implementing a reader to use in an xarray environment. If someone advanced on this we could work together. I've attached a data sample and the tables they sent me. Thanks in advance!

[202304090000.gz](https://github.com/openradar/xradar/files/11918650/202304090000.gz)
[tables_bufr_281.zip](https://github.com/openradar/xradar/files/11918651/tables_bufr_281.zip)

kmuehlbauer, 2023-07-03:
No, there haven't been any advances here. The major pain point is, and this is a show stopper, that the provide tables are in some ".csv" format. `eccodes` does not read these tables per default and they will have to be converted for that purpose to the eccodes table format.

Please ask your table provider to provide the tables in ECMWF `eccodes` format.

Francesco-Uboldi, 2023-12-13:
I am having a similar problem with some sample BUFR files I had from Météo-France. 
[](https://openradar.discourse.group/t/compute-cappi-from-meteo-france-bufr-ppi/368)

kmuehlbauer, 2023-12-13:
@Francesco-Uboldi Are your MF tables the same as linked above by @jfigui and @rzambranap?

And would you join a team to implement an `eccodes` based BUFR reader for xradar?

Francesco-Uboldi, 2023-12-13:
I have version 279 instead of 281... I will try the 281 from @rzambranap .
I can try and help, but I am quite new to both radar data in general and BUFR format. In particular, I never used `eccodes`: I met this software recently, of course... but the table problem stopped me, and I went towards OPERA instead.
By the way...  wouldn't it be nice if some BUFR user from Météo-France agreed to be involved?!?  Anyone out there?

kmuehlbauer, 2023-12-13:
@Francesco-Uboldi As I see the situation it all depends on the tables. We would need those tables in the format accepted by `eccodes`. Or there would need to be some transformation tool to convert the tables. It's a pity that BUFR data is so unaccessible.

rzambranap, 2023-12-13:
Hey everyone, I wasn't able to implement something on Python, but I was
able to sort it out via Python using the subprocess library, calling the
installed BUFR software locally. Two caveats: I couldn't launch it on VMs,
and there was something about 32 vs 64-bit architecture that somehow worked
on my PC but not on my office's servers. I can send you the tables
Météo-France sent to me and the functions I did in Python calling
subprocess and everything, but it's not up to the standards you might be
accustomed to...

ReallyNiceGuy, 2025-07-09:
@jfigui 
Check [pybufrkit](https://github.com/ywangd/pybufrkit) again, together with my tables [pybufrkit_tables](https://github.com/ReallyNiceGuy/pybufrkit_tables).
I managed to decode both your samples today.

kmuehlbauer, 2025-07-11:
This would need a brave volunteer who takes the lead here. @ReallyNiceGuy are you up for an adventure?

ReallyNiceGuy, 2025-07-11:
Before I take this step, I will need to acquaint myself with the code base.
Don't close this issue just yet ;)