openradar/xradar #104: Distinction between dataset variables and ancillary variables

egouden, 2023-04-03:
I cannot find a way to distinguish between the dataset variables and ancillary variables.

kmuehlbauer, 2023-04-03:
@egouden Could you please elaborate along an example? It's hard to guess what you are referring to.

egouden, 2023-04-04:
What about using attributes to store these ancillary variables?

kmuehlbauer, 2023-04-04:
Are you referring to those ancillary variables https://docs.openradarscience.org/projects/xradar/en/stable/datamodel.html#ancillary-variables?

It might not be possible as those variables are data variables in the underlying netcdf file. 

Maybe can describe the use case you have in mind and why you need this separated?

egouden, 2023-04-18:
There are actually 3 types of sweep variables:
- observed variables
- ancillary variables related to one or more observed variables
- metadata variables related to the sweep itself

Accessing only the observed variables should be a basic feature indeed. Is it not already needed in [our code](https://github.com/openradar/xradar/blob/main/xradar/io/export/odim.py#L80)?

egouden, 2025-04-22:
This has been solved by https://github.com/openradar/xradar/pull/257

closed 2025-04-22 by egouden