openradar/xradar #105: radar task name in sigmet files

aladinor, 2023-04-07:
* xradar version: '0.1.0'
* Python version: 3.9
* Operating System: Centos 7.9

Hi all, 

I am trying to find the task name from a Sigmet file using `xd.io.open_iris_datatree` but I can find it. is the task name included somewhere in the datatree object?

Cheers, 

Alfonso

mgrover1, 2023-04-07:
@aladinor can you post the full import + script here?

kmuehlbauer, 2023-04-07:
@aladinor Currently the backends only cover for the minimal needed data/metadata. The task name is not available, yet.

We would need to discuss if that would be something which we can map to a standard attribute, or if we would have to invent some non-standard item (possibly in a subgroup).

It would be natural to able to get all metadata from source files, so we should try to implement it.

The only backend which maps all metadata is cfradial1, because everything is described by the standard.

closed 2023-05-09 by aladinor