openradar/xradar #124: Cf/Radial1 writer

syedhamidali, 2023-08-27:
Based on the discussions held during the AMS developers meeting, there is a perceived need for a `to_cfradial1` function in the Cf/Radial1 format. The decision of its future relevance may depend on Radar Community's compatibility with Cf/Radial1. If necessary, we can consider deprecating it later. I'm eager to collaborate with fellow developers to contribute to this function, not only for its immediate purpose but also to gain experience in contributing to other functions down the line.

kmuehlbauer, 2023-08-27:
Should all Cfradial1 versions be supported? Or should we concentrate on a specific version?

mgrover1, 2023-08-27:
I think it is worth supporting both... since we would have a way to validate v1 vs. v2? Open to thought here.

syedhamidali, 2023-08-28:
Hello @kmuehlbauer 👋🏻
Which ever version is supported by the majority of the softwares out there, may be we could use the one that is in pyart.io.write

kmuehlbauer, 2023-08-28:
Hi @syedhamidali, it might be a good idea to stay in xarray-land also with the transform into CfRadial1.

So as a first proof of concept we could try to reverse the group-extracting we have in https://docs.openradarscience.org/projects/xradar/en/stable/notebooks/CfRadial1_Model_Transformation.html to recreate the source CfRadial1-file from the CfRadial2/FM301-file (current state of affairs).

While doing this we might get insights on how to treat data with different number of range_gates and even different gate spacing.

syedhamidali, 2023-08-28:
You are right! I think you've already done most of the work on it in the backends. Can't say anything about treating the data with different numbers of range_gates and even different gate spacing, may be keeping the spacing constant like in `RadXConvert: "-const_ngates"`

PR #126 (2023-09-27): Cf/Radial1 writer

- [x] Closes #124 (Preliminary)
- [x] Tests added
- [x] Changes are documented in `history.md`

I have incorporated the CFRadial1 writer into the code and tested it with four different CFRadial1 data files. It successfully processed three of the files. While it currently works with full radar volumes, I acknowledge that further refinement and compatibility with sweeps are necessary. I welcome any suggestions for improvement and am open to any modifications that enhance its suitability.

I also ran `pre-commit run --all-files`.

Thanks
Hamid
closed 2023-09-27 by kmuehlbauer