openradar/xradar #130: Add xradar team

kmuehlbauer, 2023-09-27:
This is more an information issue. But we can start a discussion from it.

I've created an `xradar`-team in the `openradar`-org and have added @mgrover1, @zssherman and @kmuehlbauer to it. I've invited @egouden, @aladinor and @syedhamidali to join.

The team has currently `triage` status. But we can discuss to level up to `write` status, if everyone is OK with it.

We already acknowledge our contributors in several ways, but we should also try to give something back in terms of responsibility. Triaging issues and pullrequest beside active code contribution is also something which will help the xradar-project grow.

mgrover1, 2023-09-27:
This sounds great @kmuehlbauer - should we have them respond to this issue to confirm they are willing to be reviewers of PRs?

kmuehlbauer, 2023-09-27:
> This sounds great @kmuehlbauer - should we have them respond to this issue to confirm they are willing to be reviewers of PRs?

Yes, might be a good idea, if they respond here.

zssherman, 2023-09-27:
@kmuehlbauer Also sounds good to me! Thanks!

kmuehlbauer, 2024-02-23:
done, see -> https://github.com/orgs/openradar/teams/xradar

closed 2024-02-23 by kmuehlbauer