openradar/xradar #135: Need some support for gridded radar data.

rcjackson, 2023-10-18:
Hello,

I have been chatting with @mgrover1 about adding some support for a standard grid object into xradar. Since the Cf/Radial format is (1) easily integrable with xarray and (2) used by most radar software packages we were siding towards making the xradar grid object to be based on the Cf/Radial standard. In addition, there is some functionality in Py-ART that would make integration with packages such as PyDDA easier, namely adding in accessors that calculate point_x, point_y, point_x, point_latitutde, point_longitude, and point_altitude from gridded Cf/Radial data. Eventually, I see the goal is for PyART/PyDDA to simply use xradar Datasets for grids, so integrating the basic routines from Py-ART grids that do not involve interpolation into xradar as a priority. I will be working on this in the coming weeks.

Bobby

kmuehlbauer, 2023-10-19:
@rcjackson Great initiative here! It would be nice to link the Cf/Radial grid standard here for interested folks.

syedhamidali, 2026-10-03:
This was implemented downstream in Py-ART shortly after this issue was opened: ARM-DOE/pyart#1479 (`pyart.xradar.Xgrid`, a CF xarray grid with `point_x/y/z`, `point_latitude/longitude/altitude`), with gridding straight from xradar objects in ARM-DOE/pyart#1468 and ARM-DOE/pyart#1719.

Together with #304 (*"Gridding is not on the roadmap for xradar (at least for now)"*) and the README note on polar-only data, I think this can be closed here.

Instead of a grid model in xradar, we could improve the xradar ↔ Py-ART link for gridded data, e.g. an xradar docs example going `DataTree` → Py-ART gridding → `Xgrid`/`xarray.Dataset`, and pointing users there from the README/docs, so the "how do I grid this?" question (#304) has a ready answer. Happy to put that together.

@rcjackson @mgrover1 @kmuehlbauer any objections?

kmuehlbauer, 2026-10-03:
@syedhamidali Yes, improving docs first is a good idea. Also over at wradlib I made 2D and 2.5D interpolation a bit more accessible. This for polar to cartesian and cartesian to polar.