openradar/xradar #149: Migrate datatree imports to xarray

mgrover1, 2024-01-25:
* xradar version: dev
* Python version: All
* Operating System: All

### Description

With datatree being deprecated soon (moving into the core xarray library), we need to change any imports to pull from xarray instead of the xarray-datatree package

https://github.com/xarray-contrib/datatree

### What I Did

I saw this on the xarray-datatree readme.

zssherman, 2024-01-25:
@mgrover1 I can always tackle this, did you all want a try and except for backwards compatibility at the moment or straight to the new xarray import?

kmuehlbauer, 2024-01-25:
Let's wait until this has settled in xarray finally. The process is currently ongoing. We should pin datatree to the last available version and make the move if  datatree has landed as full class citizen in xarray

zssherman, 2024-01-25:
Sounds good!

mgrover1, 2024-01-25:
I agree with @kmuehlbauer here! Mainly just wanted to bring it to everyone's attention :)

kmuehlbauer, 2024-10-10:
@openradar/xradar Heads-up, xarray is in full speed to complete the merge of datatree into the xarray codebase. We should act now to have everything in place when the next xarray release is out.

I'll start with adding an xarray-nightly CI run to see what we'll have to do.

eoydvin, 2024-10-25:
You might  be aware, but /xradar/accessors.py:27 (import datatree as dt)  now raises an DeprecationWarning. Using xarray v2024.10.0 and xarray-datatree v0.0.15.

kmuehlbauer, 2024-10-25:
@eoydvin Yes, we are aware. We have the xarray-nightly branch, which should work with xarray 2024.10.0.

cchwala, 2024-10-25:
Thanks for the info @kmuehlbauer. Can you estimate when there will be a release on pypi that drop the dependency on datatree?

kmuehlbauer, 2024-10-25:
ASAP, next Monday latest.

kmuehlbauer, 2024-10-25:
So, to make the transition as smooth as possible, we should:

1. pin xarray to <=2024.9.0 and do a release 0.7.0
2. merge #218, pin xarray to >=2024.10.0 and do another release 0.8.0

PR #213 (2024-10-16): Add xarray nightly CI run

- [x] Closes #149
- [x] Tests added/changes
- [ ] Changes are documented in `history.md`
- [ ] Docs changed accordingly
closed 2024-10-28 by kmuehlbauer