openradar/xradar #163: ENH: Bring in extended list of supported message structures from MetPy for Level2 reader

jthielen, 2024-03-15:
As referenced in https://github.com/openradar/xradar/pull/158 and discussed with @mgrover1 in the MetPy telecon yesterday, there is still some relevant code in https://github.com/openradar/xradar/compare/main...jthielen:xradar:nexrad-level2 that may be helpful, namely that of a reconciled list of message structures combining those from both Py-ART's and MetPy's. Given that prior effort used MetPy's bespoke `NamedStruct` and `BitField` objects to handle this, there would likely be a quite a bit of refactoring needed (going either direction..modifying the reconciled lists to match the current xradar approach, or updating the xradar approach to use these MetPy-style objects). This issue can hopefully track this particular facet of the overall effort of https://github.com/openradar/xradar/issues/40.

kmuehlbauer, 2024-03-15:
Thanks @jthielen for pushing this here.

I can have a look what would be necessary to incorporate things like `BitField` into the current structure. 

I'm also interested to check if and how `NamedStruct` can be utilized for the current processing.

wx4stg, 2024-10-28:
I'll just re-up this as there's a particular proprietary radar file that claims to be "level 2 format" (I can't share sadly, and I don't have time to look into it right now... I'm being soooo helpful, I know :/ ) that opens in metpy and pyart but does not open in xradar...