openradar/xradar #164: AttributeError: 'DataTree' object has no attribute 'sweep_group_name'

aladinor, 2024-03-24:
* xradar version:  0.4.3
* Python version: 3.12
* Operating System: Ubuntu

### Description

Hi every one, 

I've tried to reproduce this [grinding](https://arm-doe.github.io/pyart/examples/xradar/plot_grid_xradar.html) example using sigmet files available in amazon. 

```python 
import xradar as xd
import fsspec
from datetime import datetime
import pyart
import matplotlib.pyplot as plt

def create_query(date, radar_site):
    """
    Creates a string for quering the IDEAM radar files stored in AWS bucket
    :param date: date to be queried. e.g datetime(2021, 10, 3, 12). Datetime python object
    :param radar_site: radar site e.g. Guaviare
    :return: string with a IDEAM radar bucket format
    """
    if (date.hour != 0) and (date.minute != 0):
        return f"l2_data/{date:%Y}/{date:%m}/{date:%d}/{radar_site}/{radar_site[:3].upper()}{date:%y%m%d%H%M}"
    elif (date.hour != 0) and (date.minute == 0):
        return f"l2_data/{date:%Y}/{date:%m}/{date:%d}/{radar_site}/{radar_site[:3].upper()}{date:%y%m%d%H}"
    else:
        return f"l2_data/{date:%Y}/{date:%m}/{date:%d}/{radar_site}/{radar_site[:3].upper()}{date:%y%m%d}"

def main():
    date_query = datetime(2022, 10, 6)
    radar_name = "Guaviare"
    query = create_query(date=date_query, radar_site=radar_name)
    str_bucket = "s3://s3-radaresideam/"
    fs = fsspec.filesystem("s3", anon=True)
    radar_files = sorted(fs.glob(f"{str_bucket}{query}*"))
    file = fsspec.open_local(
        f"filecache::s3://{radar_files[0]}",
        s3={"anon": True},
        filecache={"cache_storage": "/tmp/radar"},
    )

    tree1 = xd.io.open_iris_datatree(file)
    radar = pyart.xradar.Xradar(tree1, default_sweep='sweep_0')
    # Grid using 11 vertical levels, and 101 horizontal grid cells at a resolution on 1 km
    grid = pyart.map.grid_from_radars(
        (radar,),
        grid_shape=(1, 501, 501),
        grid_limits=(
            (0.0, 10_000),
            (-300_000.0, 300_000.0),
            (-300_000, 300_000.0),
        ),
    )

    display = pyart.graph.GridMapDisplay(grid)
    display.plot_grid(
        "DBZH", level=0, vmin=-20, vmax=60, cmap="pyart_ChaseSpectral"
    )
    plt.show()
    print(1)

if __name__ == "__main__":
    main()
```
However, I got the following error, 

```python 
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[6], line 1
----> 1 radar = pyart.xradar.Xradar(tree1, default_sweep='sweep_0')

File [~/miniconda3/envs/raw2zarr/lib/python3.12/site-packages/pyart/xradar/accessor.py:274](http://127.0.0.1:7022/~/miniconda3/envs/raw2zarr/lib/python3.12/site-packages/pyart/xradar/accessor.py#line=273), in Xradar.__init__(self, xradar, default_sweep, scan_type)
    272 self.xradar = xradar
    273 self.scan_type = scan_type or "ppi"
--> 274 self.combined_sweeps = self._combine_sweeps(self.xradar)
    275 self.fields = self._find_fields(self.combined_sweeps)
    276 self.time = dict(
    277     data=(self.combined_sweeps.time - self.combined_sweeps.time.min()).astype(
    278         "int64"
   (...)
    282     calendar="gregorian",
    283 )

File [~/miniconda3/envs/raw2zarr/lib/python3.12/site-packages/pyart/xradar/accessor.py:552](http://127.0.0.1:7022/~/miniconda3/envs/raw2zarr/lib/python3.12/site-packages/pyart/xradar/accessor.py#line=551), in Xradar._combine_sweeps(self, radar)
    549 def _combine_sweeps(self, radar):
    550     # Loop through and extract the different datasets
    551     ds_list = []
--> 552     for sweep in radar.sweep_group_name.values:
    553         ds_list.append(radar[sweep].ds.drop_duplicates("azimuth"))
    555     # Merge based on the sweep number

File [~/miniconda3/envs/raw2zarr/lib/python3.12/site-packages/datatree/common.py:54](http://127.0.0.1:7022/~/miniconda3/envs/raw2zarr/lib/python3.12/site-packages/datatree/common.py#line=53), in TreeAttrAccessMixin.__getattr__(self, name)
     52         with suppress(KeyError):
     53             return source[name]
---> 54 raise AttributeError(
     55     f"{type(self).__name__!r} object has no attribute {name!r}"
     56 )

AttributeError: 'DataTree' object has no attribute 'sweep_group_name'
```
Digging around, I found out that the `xd.io.open_iris_datatree` data loader is not adding the `sweep_group_name`, `sweep_fixed_angle`, and `radar_parameters` to the datatree. I worked around and I submitted a push request where this issue is solved following the FM301 standard WMO data model

kmuehlbauer, 2024-03-25:
I see now, that xradar actually defines `sweep_group_name` and `sweep_fixed_angle` in `xradar.model.required_root_vars`. This might be due to inconsistency between the `CFRadialDoc-v2.1-20190901` document in https://github.com/NCAR/CfRadial/tree/master/docs and the above linked official WMO document. 

@mgrover1 How should we move on here? Removing these two variables from `required_root_vars` seems reasonable. Also re-checking mandatory, conditional and optional variables/attributes seems necessary now.

mgrover1, 2024-03-25:
Let's go ahead and remove `sweep_group_name` and `sweep_fixed_angle` from `required_root_vars`!

aladinor, 2024-03-25:
Based on this, I will remove `sweep_group_name` and `sweep_fixed_angle` from the #165 pr.

mgrover1, 2024-03-25:
Thanks for raising this issue @aladinor  - I will work to update the Py-ART side to fix this as well.

mgrover1, 2024-03-28:
@aladinor - with #167 merged, a new release for xradar, and the following PR for Py-ART
https://github.com/ARM-DOE/pyart/pull/1537

This issue should be resolved; I tested it with existing ODIM_H5 files that would have previously not worked, that work now. Feel free to give that branch a try - hoping to merge + cut a minor release of Py-ART for you to try out. Thanks again for raising this issue.

PR #166 (2024-03-29): adding radar parameters to xradar iris datatree
adding radar parameters to xradar iris datatree

- [x] Closes #164
closed 2024-03-29 by mgrover1