openradar/xradar #170: missing get_azimuth method in obj created from xradar

seoc16, 2024-03-29:
* xradar version: 0.4.3
* Python version: 3.12.2
* Operating System: linux

### Description

Describe what you were trying to get done.
Tell us what happened, what went wrong, and what you expected to happen.
i am loading data through xradar using 
```python
radar = pyart.xradar.Xradar(tree1,  default_sweep='sweep_0')
```
I am able to use the gridding function in pyart,  but then i am trying to use the find objects method and i get the following error
```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[123], line 2
      1 #attempt at tracking obj
----> 2 objDisplay = pyart.correct.find_objects(radar, "DBZH", 15,sweeps=0)

File ~/miniconda3/envs/radar/lib/python3.12/site-packages/arm_pyart-1.18.0.post5-py3.12-linux-x86_64.egg/pyart/correct/despeckle.py:70, in find_objects(radar, field, threshold, sweeps, smooth, gatefilter, delta)
     68 for iswp in sweeps:
     69     data = _get_data(radar, iswp, field, tlo, thi, smooth, gatefilter=gatefilter)
---> 70     az = radar.get_azimuth(iswp, copy=False)
     71     if _check_for_360(az, delta):
     72         # If 360 or close, account for the periodic boundary
     73         labels, nobj = _adjust_for_periodic_boundary(data)

AttributeError: 'Xradar' object has no attribute 'get_azimuth'
```
are there missing methods in the radar object?

### What I Did
```python
objDisplay = pyart.correct.find_objects(radar, "DBZH", 15,sweeps=0)
```

zssherman, 2024-03-29:
@seoc16 I believe the despeckling algo in Py-ART is not yet able to use an xradar object. The call get_azimuth being used in the find_objects code above is found in a Py-ART radar class object. @mgrover1 Can correct me if I'm wrong, or if there is a status on that capability.

mgrover1, 2024-04-01:
Thanks for raising this issue - this is a bug within the Py-ART/xradar bridge currently.

mgrover1, 2024-04-01:
@seoc16 - please see https://github.com/ARM-DOE/pyart/issues/1535 for more discussion about resolving this issue 👍

kmuehlbauer, 2024-04-22:
Should be resolved by updating pyart to latest version.

closed 2024-04-22 by kmuehlbauer