openradar/xradar #171: open_datatree_from_Zarr (?)

aladinor, 2024-04-10:
* xradar version: 
* Python version:3.12
* Operating System:Ubunut 22.04

### Description

Hi everyone, 

I was wondering if we should create an `io` method that allows us to read `zarr` back to `xradar`. Something like `xradar.io.open_zarr1` given that we can now store/save the `xradar` hierarchical structure object using `zarr`.

Please let me know your thoughts.

kmuehlbauer, 2024-04-12:
@aladinor doesn't this work out-of-the-box with `datatree`? If I'm reading the documentation https://xarray-datatree.readthedocs.io/en/latest/io.html?highlight=zarr#zarr this should just work, no?

There is also work currently to integrate `datatree` into xarray pushed by NASA, NSIDC:
https://github.com/pydata/xarray/projects/9

There is a weekly open dev meeting for that purpose, if you are interested:
https://github.com/pydata/xarray/issues/8747

But good chance I'm missing something, so please give a bit more detail.

aladinor, 2024-04-12:
Hi @kmuehlbauer. Well, I was thinking of a wrapper that uses `datatree` functionality to read Zarr files back to `xradar`. Currently, if we want to read data, we have to use `datatree.open_datatre`. I was just thinking of something like `xradar.io.datatree` and then passing the `engine='zarr' keyword. I wasn't thinking about a whole development. Sorry for the misunderstanding. 

Let me know your thoughts

Steve-Roderick, 2025-02-18:
Just in case, it's worth noting that `xradar` appears to be tested against `zarr` 2.x versions and not `zarr` 3.x versions. Based on a little experience, `zarr` 3.x is not really backwards compatible with 2.x. Something to keep in mind.

syedhamidali, 2026-10-03:
@aladinor I think this is covered by xarray itself now, as @kmuehlbauer suggested: with xarray 2026.9 and `zarr>=3` (#398), an xradar tree round-trips through Zarr without a wrapper:

```python
dtree = xd.io.open_odim_datatree(DATASETS.fetch("71_20181220_060628.pvol.h5"))
dtree.to_zarr("vol.zarr")
dtree2 = xr.open_datatree("vol.zarr", engine="zarr")
dtree2["sweep_0"].ds.identical(dtree["sweep_0"].ds)   # True
dtree2.xradar.georeference()                           # works
```

(see [xarray DataTree I/O](https://docs.xarray.dev/en/stable/user-guide/io.html#zarr)). @Steve-Roderick's zarr 3 concern is handled by the `zarr>=3` pin. OK to close?

kmuehlbauer, 2026-10-15:
I'm going to close this, @aladinor. Please open a new issue, if need be.

closed 2026-10-15 by kmuehlbauer