openradar/xradar #189: Open Data using File-Like Object

Ockenfuss, 2024-08-06:
* xradar version: 0.6.0
* Python version: 3.11.5
* Operating System: Alma Linux

Is it possible to use a file-like object instead of filenames when using xradar? According to the xarray open_dataset documentation, as well as the [xradar Odim Backend Documentation](https://docs.openradarscience.org/projects/xradar/en/stable/generated/xradar.io.backends.odim.OdimBackendEntrypoint.html#xradar.io.backends.odim.OdimBackendEntrypoint), file objects are supported, but with the 'odim' engine, I get an error:
```python
from open_radar_data import DATASETS
import xarray as xr
import xradar as xd
filename = DATASETS.fetch("71_20181220_060628.pvol.h5")
#works:
ds = xr.open_dataset(filename, group="sweep_0", engine="odim") 
#works as well (at least no exception raised):
with open(filename, 'rb') as f:
    ds = xr.open_dataset(f)
# Raises TypeError: cannot pickle '_io.BufferedReader' object
with open(filename, 'rb') as f:
    ds = xr.open_dataset(f, group="sweep_0", engine="odim")
```

Thank you for any help and the xradar development in general!
Paul

kmuehlbauer, 2024-08-06:
@Ockenfuss 

Thanks for raising and the clear description. Could you please also provide the full traceback? Thanks!

Which xarray version are you on?

Ockenfuss, 2024-08-06:
I am on the current xarray version 2024.7.0. Can you reproduce it?

<details>
  <summary>Traceback</summary>

```python
Traceback (most recent call last):
  File "/scratch/p/Paul.Ockenfuss/Test/test_polardata/xradar_bug.py", line 6, in <module>
    ds = xr.open_dataset(f, group="sweep_0", engine="odim")
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/scratch-local/Paul.Ockenfuss/virtual_environments/radar/lib/python3.11/site-packages/xarray/backends/api.py", line 588, in open_dataset
    backend_ds = backend.open_dataset(
                 ^^^^^^^^^^^^^^^^^^^^^
  File "/scratch-local/Paul.Ockenfuss/virtual_environments/radar/lib/python3.11/site-packages/xradar/io/backends/odim.py", line 825, in open_dataset
    ds = ds.sortby(dim0)
         ^^^^^^^^^^^^^^^
  File "/scratch-local/Paul.Ockenfuss/virtual_environments/radar/lib/python3.11/site-packages/xarray/core/dataset.py", line 8182, in sortby
    aligned_vars = align(self, *arrays, join="left")
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/scratch-local/Paul.Ockenfuss/virtual_environments/radar/lib/python3.11/site-packages/xarray/core/alignment.py", line 882, in align
    aligner.align()
  File "/scratch-local/Paul.Ockenfuss/virtual_environments/radar/lib/python3.11/site-packages/xarray/core/alignment.py", line 582, in align
    self.reindex_all()
  File "/scratch-local/Paul.Ockenfuss/virtual_environments/radar/lib/python3.11/site-packages/xarray/core/alignment.py", line 557, in reindex_all
    self.results = tuple(
                   ^^^^^^
  File "/scratch-local/Paul.Ockenfuss/virtual_environments/radar/lib/python3.11/site-packages/xarray/core/alignment.py", line 558, in <genexpr>
    self._reindex_one(obj, matching_indexes)
  File "/scratch-local/Paul.Ockenfuss/virtual_environments/radar/lib/python3.11/site-packages/xarray/core/alignment.py", line 546, in _reindex_one
    return obj._reindex_callback(
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/scratch-local/Paul.Ockenfuss/virtual_environments/radar/lib/python3.11/site-packages/xarray/core/dataset.py", line 3517, in _reindex_callback
    reindexed = self.copy(deep=aligner.copy)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/scratch-local/Paul.Ockenfuss/virtual_environments/radar/lib/python3.11/site-packages/xarray/core/dataset.py", line 1366, in copy
    return self._copy(deep=deep, data=data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/scratch-local/Paul.Ockenfuss/virtual_environments/radar/lib/python3.11/site-packages/xarray/core/dataset.py", line 1402, in _copy
    variables[k] = v._copy(deep=deep, data=data.get(k), memo=memo)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/scratch-local/Paul.Ockenfuss/virtual_environments/radar/lib/python3.11/site-packages/xarray/core/variable.py", line 925, in _copy
    ndata = copy.deepcopy(ndata, memo)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 172, in deepcopy
    y = _reconstruct(x, memo, *rv)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 271, in _reconstruct
    state = deepcopy(state, memo)
            ^^^^^^^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 146, in deepcopy
    y = copier(x, memo)
        ^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 211, in _deepcopy_tuple
    y = [deepcopy(a, memo) for a in x]
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 211, in <listcomp>
    y = [deepcopy(a, memo) for a in x]
         ^^^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 146, in deepcopy
    y = copier(x, memo)
        ^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 231, in _deepcopy_dict
    y[deepcopy(key, memo)] = deepcopy(value, memo)
                             ^^^^^^^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 172, in deepcopy
    y = _reconstruct(x, memo, *rv)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 271, in _reconstruct
    state = deepcopy(state, memo)
            ^^^^^^^^^^^^^^^^^^^^^
[... 78 lines cut ...]
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 146, in deepcopy
    y = copier(x, memo)
        ^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 211, in _deepcopy_tuple
    y = [deepcopy(a, memo) for a in x]
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 211, in <listcomp>
    y = [deepcopy(a, memo) for a in x]
         ^^^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 146, in deepcopy
    y = copier(x, memo)
        ^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 211, in _deepcopy_tuple
    y = [deepcopy(a, memo) for a in x]
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 211, in <listcomp>
    y = [deepcopy(a, memo) for a in x]
         ^^^^^^^^^^^^^^^^^
  File "/software/opt/el_9/x86_64/python/3.11-2023.09/lib/python3.11/copy.py", line 161, in deepcopy
    rv = reductor(4)
         ^^^^^^^^^^^
TypeError: cannot pickle '_io.BufferedReader' object
```

</details>

kmuehlbauer, 2024-08-07:
Yes, and I think I've already hit this, if you look at the tests:

https://github.com/openradar/xradar/blob/bf03ab86085d8e86a2407bd33a5214c8a605202f/tests/io/test_io.py#L222-L232

The issue is triggered by `ds.sortby()`, we can strip this down to pure xarray:

```python
import numpy as np
import netCDF4
import xarray as xr
import fsspec
import io

# create file with netCDF4 to have all bits and pieces
filename = "test.h5"
with netCDF4.Dataset(filename,'w') as f:
    f.createDimension('x', 3)
    f.createDimension('y', 6)
    var = f.createVariable('var', 'int8', ('x', 'y'))
    xcoord = f.createVariable('x', 'int8', ('x'))
    ycoord = f.createVariable('y', 'int8', ('y'))
    xcoord[:] = [2, 1, 0]
    ycoord[:] = [0, 1, 2, 3, 4, 5]
    var[:] = np.arange(18).reshape((3, 6))
```

This works, maybe because `CachingFileManager` ist used?

```python
with xr.backends.H5NetCDFStore.open(filename) as store:
    store_entrypoint = xr.backends.StoreBackendEntrypoint()
    ds = store_entrypoint.open_dataset(store)
    display(ds["var"].values)
    ds = ds.sortby('x')
    display(ds["var"].values)
```
```
array([[ 0,  1,  2,  3,  4,  5],
       [ 6,  7,  8,  9, 10, 11],
       [12, 13, 14, 15, 16, 17]], dtype=int8)

array([[12, 13, 14, 15, 16, 17],
       [ 6,  7,  8,  9, 10, 11],
       [ 0,  1,  2,  3,  4,  5]], dtype=int8)
```
This works too, with `reindex`.

```python
with open(filename, 'rb') as f:
    print(type(f))
    store = xr.backends.H5NetCDFStore.open(f)
    store_entrypoint = xr.backends.StoreBackendEntrypoint()
    ds = store_entrypoint.open_dataset(store)
    display(ds["var"].values)
    ds = ds.reindex({"x": np.array([0, 1, 2])})
    display(ds["var"].values)
```
```python
<class '_io.BufferedReader'>
array([[ 0,  1,  2,  3,  4,  5],
       [ 6,  7,  8,  9, 10, 11],
       [12, 13, 14, 15, 16, 17]], dtype=int8)

array([[12, 13, 14, 15, 16, 17],
       [ 6,  7,  8,  9, 10, 11],
       [ 0,  1,  2,  3,  4,  5]], dtype=int8)
```

This breaks with sortby.

[... 283 lines cut ...]
    200 def _deepcopy_tuple(x, memo, deepcopy=deepcopy):
--> 201     y = [deepcopy(a, memo) for a in x]
    202     # We're not going to put the tuple in the memo, but it's still important we
    203     # check for it, in case the tuple contains recursive mutable structures.
    204     try:

File [/home/kai/data/mambaforge/envs/xr_312_np2/lib/python3.12/copy.py:151](http://localhost:8888/lab/tree/home/kai/python/gists/xradar/home/kai/data/mambaforge/envs/xr_312_np2/lib/python3.12/copy.py#line=150), in deepcopy(x, memo, _nil)
    149 reductor = getattr(x, "__reduce_ex__", None)
    150 if reductor is not None:
--> 151     rv = reductor(4)
    152 else:
    153     reductor = getattr(x, "__reduce__", None)

TypeError: cannot pickle 'BufferedReader' instances
```
</details>

There are quite a bunch of GH issues out there, which are discussing that problem. It seems, that [io.BufferedReader](https://docs.python.org/3/library/io.html#io.BufferedReader) can't be pickled. 

XRef: 
- https://github.com/fsspec/filesystem_spec/issues/579
- https://github.com/pauldmccarthy/indexed_gzip/issues/125
- tbc

What I'm wondering is why the reindex above works and the sortby fails? @benbovy, can you shed some light here?

kmuehlbauer, 2024-08-07:
> Interesting that it works with reindex, but not sortby.

There is a deepcopy in the sortby code flow somewhere, but not in the reindex. Let's see if we get some answers from @benbovy (sorry for tagging twice).

benbovy, 2024-08-12:
Sorry @kmuehlbauer not sure what is causing the different behavior here, although I haven't inspected further the reindex vs. sortby code.

kmuehlbauer, 2024-08-12:
Thanks @benbovy, I'll open an issue on the xarray tracker for further investigation.