openradar/xradar #212: DOC: Add an example working with uxarray

mgrover1, 2024-09-27:
* xradar version: All
* Python version: All
* Operating System: All

### Description

Add an example fitting xradar datasets/datatrees into uxarray, properly representing the gate centers and edges. This could be accompanied by an accessor that transform the datatree/dataset for the user as well 👍 

This will make the downstream functions in wradlib/py-art much easier to maintain, and allows us to take advantage of some neat functionality on the uxarray side

see related issue here https://github.com/UXARRAY/uxarray/issues/976

kmuehlbauer, 2024-09-27:
@mgrover1 Looking forward to it.

syedhamidali, 2026-10-04:
I've prototyped this in radarx: `to_uxarray()` turns a PPI sweep into a uxarray dataset with one quadrilateral face per gate. The corner nodes lie halfway to the neighbouring rays and gates, full 360° sweeps close on themselves, and faces are counter-clockwise as UGRID requires.

```python
import radarx  # registers the .radarx accessor

sweep = dtree["sweep_0"].to_dataset(inherit="all_coords")
uxds = sweep.radarx.to_uxarray(["DBZH"])
```

Checks on a NEXRAD sweep (720 × 600 gates):
- The conversion takes 0.2 s.
- Face areas match the analytic gate area Δr · r · Δφ to within 0.03 %.
- Face centres are within about 2 m (median) of xradar's georeferenced gate centres.

The notebook compares the native polar sweep, a Cartesian grid and the UGRID grid side by side, near and far from the radar. It also shows area-weighted statistics and remapping to a lat/lon grid: https://radarx.readthedocs.io/en/latest/notebooks/Radar_UGRID_uxarray.html

Code: https://github.com/syedhamidali/radarx/blob/main/radarx/grid/ugrid.py. Happy to move it into xradar as an accessor if that's the preferred home.

kmuehlbauer, 2026-10-15:
@syedhamidali This looks very nice. There is also a GDAL-based implementation in wradlib using shapefiles. I'll have a look there again when I have the time, if I can connect to UGRID somehow.