openradar/xradar #217: Implement functionality to test if a dataset contains a sweep

kmuehlbauer, 2024-10-17:
* xradar version: latest
* Python version: any
* Operating System: any

### Description

In #203 a check is added if a `xarray.Dataset` contains a radar sweep. Currently the test is to check for existing `range`-dimension (which works now, but might be a bit fragile). I'm proposing to implement kind of `xradar.is_sweep()`/`ds.xradar.is_sweep()` functionality. This is connected to the idea of schema-validation (see #31).

syedhamidali, 2026-10-03:
Status on `main`: there's no `is_sweep` yet, and the `range` check from #203 is gone. Sweeps are now picked by name, in three slightly different ways:

- `util.get_sweep_keys` accepts `sweep_<int>` ([util.py#L507-L539](https://github.com/openradar/xradar/blob/21c034c/xradar/util.py#L507-L539))
- `apply_to_sweeps` / `map_over_sweeps` use `match("sweep*")` + `startswith("/sweep")` ([util.py#L586-L587](https://github.com/openradar/xradar/blob/21c034c/xradar/util.py#L586-L587))
- the ODIM writer uses `match("sweep_*")` ([odim.py#L188](https://github.com/openradar/xradar/blob/21c034c/xradar/io/export/odim.py#L188))

The `map_over_sweeps` docstring still describes the old check (*"checks whether the dataset provided to the function has the 'range' dimension"*, [util.py#L624](https://github.com/openradar/xradar/blob/21c034c/xradar/util.py#L624)).

Proposal: `xd.util.is_sweep(ds, strict=False)` + `ds.xradar.is_sweep()`:
- default: a `range` dim plus one of `time`/`azimuth`/`elevation` as the ray dim (structure, not name)
- `strict=True`: also requires `model.required_sweep_metadata_vars` ([model.py#L158-L166](https://github.com/openradar/xradar/blob/21c034c/xradar/model.py#L158-L166)), which mirrors FM301 Table 301-7a ([WMO-No. 306, Vol. I.2](https://library.wmo.int/idurl/4/35625)): `sweep_number`, `sweep_mode`, `follow_mode`, `prt_mode`, `sweep_fixed_angle`, `azimuth`, `elevation`. That would be a first small step towards #31.

`get_sweep_keys` / `apply_to_sweeps` could then use it instead of name matching. @kmuehlbauer OK to go ahead, or would you rather wait for #31?

kmuehlbauer, 2026-10-03:
Implementation of the pandera based validation would be nice. But quickly checking if a sweep is a sweep, even if not fully conforming to standard would be reasonable, too.

closed 2026-10-07 by kmuehlbauer