openradar/xradar #223: xradar release 0.8.0

kmuehlbauer, 2024-10-28:
We've released [xradar 0.7.0](https://github.com/openradar/xradar/releases/tag/v0.7.0) as last version using https://github.com/xarray-contrib/datatree and `xarray <= 2024.9.0`.

With the merge of #218 we are prepared to release `xradar 0.8.0` and import DataTree from xarray with `xarray <= 2024.10.0`.

Before this release there are at least these two PR's which should get merged:

- [ ] #203 
- [ ] #221

aladinor, 2024-10-28:
I closed #221 since I will wait for the new 0.8.0 release and add all changes I have regarding `from_dict` including the `radar_parameters` and the `calibration` groups.

kmuehlbauer, 2024-10-28:
@aladinor Wouldn't it make more sense to have 0.8.0 include the best starting experience with the xarray based datatree?

aladinor, 2024-10-28:
I was thinking about it because I didn't want to delay the release since I don't have my laptop near me. However, I can reopen it later today and submit all new changes

kmuehlbauer, 2024-10-28:
I didn't want to impose any pressure on you, sorry. I've set the release date to October 31st. Do you think you can manage that?

closed 2024-11-04 by kmuehlbauer