openradar/xradar #232: Missing radar_parameters and calibrations, and georeferencing in nexrad_level2

syedhamidali, 2024-11-02:
* xradar version: 0.7+
* Python version:3.12
* Operating System:MacOS

### Description
I cant see radar parameters and georeferencing groups

### What I Did

```python
import xradar as xd
from open_radar_data import DATASETS

file = DATASETS.fetch("KLBB20160601_150025_V06")

tree = xd.io.open_nexradlevel2_datatree(file)

print(tree.groups)
```

```python
('/',
 '/sweep_0',
 '/sweep_1',
 '/sweep_2',
 '/sweep_3',
 '/sweep_4',
 '/sweep_5',
 '/sweep_6',
 '/sweep_7',
 '/sweep_8',
 '/sweep_9',
 '/sweep_10')
```

aladinor, 2024-11-02:
I think this might resolved with #221.

syedhamidali, 2024-11-02:
@aladinor Yes, I installed your data-constructor branch and gave it a try, it is working... Thanks!

syedhamidali, 2024-11-02:
Solved in #221

syedhamidali, 2024-11-02:
Although @aladinor’s changes resolved the issue with these groups, it turns out they are empty. I think the error lies in the nexrad_level2 reader.

kmuehlbauer, 2024-11-03:
Currently there is no mapping between the nexrad metadata and those groups. Nexrad is not the only engine which would need a proper mapping. #221 was the first big step to make this possible at all by making these groups available (even if not filled).

We should keep this issue open until we've worked the different backends mapping.

syedhamidali, 2026-10-03:
Status check on `main` (open-radar-data samples, `optional_groups=True`), number of variables per group:

| backend | `radar_parameters` | `radar_calibration` | `georeferencing_correction` |
|---|---|---|---|
| CfRadial1 | 5 | 45 | 16 |
| GAMIC | – | 6 | – |
| Furuno | – | 2 | – |
| NEXRAD L2 (`KLBB20160601_150025_V06`, the original example) | – | – | – |
| ODIM, IRIS, Rainbow, HPL | – | – | – |

So this should stay open. The information is in the files: ODIM `how` (`beamwH/V`, `wavelength`/`frequency`, `antgainH/V`, `radconstH/V`, see [ODIM_H5 v2.4](https://www.eumetnet.eu/wp-content/uploads/2021/07/ODIM_H5_v2.4.pdf) Table 8), the NEXRAD metadata records, and the IRIS task configuration.

I could start with ODIM, which overlaps with the ODIM 2.4 mandatory `how` attributes in #98, then NEXRAD, one backend per PR. @kmuehlbauer does that sound right?

kmuehlbauer, 2026-10-03:
We would need to go over the backends one by one and make the metadata available. As you said, several of these are already in the files, but do not get pulled out.

syedhamidali, 2026-10-03:
I plan to complete this by the end of the day, inshallah!

syedhamidali, 2026-10-03:
First backend: ODIM in #431 (beam widths, antenna gains, radar constants, NEZH/V, radome loss from `how`).

Following up with numbers on round trips: with the default `optional_groups=False`, `open_cfradial1_datatree(f)` → `to_cfradial1` keeps 1 of the 53 metadata variables of `cfrad.20080604_002217_000_SPOL_v36_SUR.nc` (all 53 with `optional_groups=True`; `to_cfradial2` on the CfRadial1 dataset keeps them all).

#333 made the groups opt-in because they were mostly empty. As the backends get filled now, could we go back to your first suggestion in #331, *"read all populated groups and skip only empty groups by default"*? Trees stay lean for backends without metadata, and conversions don't silently lose calibration.

syedhamidali, 2026-10-03:
Status: one PR per backend, as suggested. With `optional_groups=True`:

| backend | PR | `radar_parameters` | `radar_calibration` |
|---|---|---|---|
| ODIM | #431 | beam widths, antenna gains | radar constants, NEZH/V, radome loss |
| NEXRAD L2 | #433 | – (needs Message 18, #163) | transmit power, noise, system ΦDP |
| Furuno | #434 | beam widths | – (scaling of the other SCNX fields: question in the PR) |
| IRIS | #435 | beam widths, receiver bandwidth | radar constants, noise, I0 |
| GAMIC | #437 | beam widths | now floats with units and model names (were strings under GAMIC names) |
| Rainbow | #438 | beam widths, antenna gains, receiver bandwidth | radar constants, transmit power |
| UF | #439 | beam widths, receiver bandwidth (now read in MHz, as documented) | – |

Each PR maps only fields whose meaning and units are documented (ODIM, ICD, IRIS Programming Guide, UF documentation) or unambiguous from name and value, and lists the ones left out (mostly ZDR/ΦDP offsets with unclear sign convention, and dBZ0-type constants).

Nothing to map for the rest: the DataMet sample only has data scaling in `calibration.txt`, Metek MRR only its own linear `CC` constant, and the Halo `.hpl` header has no instrument parameters (FM301 `lidar_parameters` would need beam divergence, aperture, pulse energy, ...). CfRadial1 already fills all three groups.

Together with `DataTree.prune()` (xarray ≥ 2025.08), reading these groups by default and dropping the empty ones would now be a small change, see my comment above.

kmuehlbauer, 2026-10-29:
@syedhamidali Can we close this one?

syedhamidali, 2026-10-29:
@kmuehlbauer yes, I think so. Every reader that carries this metadata fills `radar_parameters` and `radar_calibration` now. Number of variables with `optional_groups=True` on current main, with the open-radar-data samples:

| backend | `radar_parameters` | `radar_calibration` |
|---|---|---|
| CfRadial1 | 5 | 45 |
| ODIM | 2 | 0 (this sample has no calibration attributes in `how`, the tests use files which have them) |
| GAMIC | 2 | 4 |
| NEXRAD Level II (`KLBB20160601_150025_V06`, the file of this issue) | 0 | 3 |
| IRIS | 3 | 5 |
| Furuno (SCNX) | 2 | 7 |
| Rainbow | 4 | 2 |
| IMD | 3 | 8 |
| UF | 2 | 0 (no calibration attributes in the sample) |
| DataMet, HPL, Metek | 0 | 0 (nothing to map) |

What is left is tracked separately:
- NEXRAD `radar_parameters` needs the Message 18 structures: #163
- one value per sweep in `radar_calibration` (`calib` dimension): #541
- vendor-specific names in the CfRadial1 export: #555
- Furuno noise power name: #538
- `georeferencing_correction`, which is only filled for CfRadial1: #568 (new, mostly a question which readers have such values)

Closing this one.

closed 2026-10-29 by syedhamidali