openradar/xradar #236: `ValueError` when reading nc files from cloud storage bucket

aladinor, 2024-11-08:
* xradar version: 0.8.0
* Python version: 3.12
* Operating System: Ubuntu 2404

Hi everyone, 

I was working with some Sigmet/Iris files and I found that I can read data from an s3 bucket using data streaming as follows,

```python 
import xradar as xd
import fsspec

fs = fsspec.filesystem('s3', anon=True)
files = sorted(fs.glob("s3-radaresideam/l2_data/2022/08/09/Carimagua/CAR22080919*"))
flist = ['s3://' + f for f in files]
u = flist[1]
with fsspec.open(u, "rb", anon=True) as file:
    dt = xd.io.open_iris_datatree(file.read())
```
And successfully got an Xarray.Datatree with all the information within the file. 

```python
<xarray.DataTree>
Group: /
│   Dimensions:              (sweep: 2)
│   Dimensions without coordinates: sweep
│   Data variables:
│       volume_number        int64 8B 0
│       platform_type        <U5 20B 'fixed'
│       instrument_type      <U5 20B 'radar'
│       time_coverage_start  <U20 80B '2022-08-09T19:03:15Z'
│       time_coverage_end    <U20 80B '2022-08-09T19:03:58Z'
│       longitude            float64 8B -71.33
│       altitude             float64 8B 206.0
│       latitude             float64 8B 4.564
│       sweep_fixed_angle    (sweep) float64 16B 6.4 8.0
│       sweep_group_name     (sweep) int64 16B 0 1
│   Attributes:
│       Conventions:      None
│       instrument_name:  carimagua-radar
│       version:          None
│       title:            None
│       institution:      None
│       references:       None
│       source:           Sigmet
│       history:          None
│       comment:          Tercer tarea del modo precipitacion / 6.4 - 8
│       scan_name:        PRECB       
├── Group: /sweep_0
│       Dimensions:            (azimuth: 360, range: 581)
│       Coordinates:
│           elevation          (azimuth) float32 1kB ...
│           time               (azimuth) datetime64[ns] 3kB 2022-08-09T19:03:16.92500...
│           longitude          float64 8B ...
│           latitude           float64 8B ...
│           altitude           float64 8B ...
│         * azimuth            (azimuth) float32 1kB 0.07416 1.159 2.079 ... 358.1 359.2
│         * range              (range) float32 2kB 1e+03 1.3e+03 ... 1.747e+05 1.75e+05
│       Data variables: (12/17)
│           DBTH               (azimuth, range) float32 837kB ...
│           DBZH               (azimuth, range) float32 837kB ...
│           VRADH              (azimuth, range) float32 837kB ...
│           WRADH              (azimuth, range) float32 837kB ...
│           ZDR                (azimuth, range) float32 837kB ...
│           KDP                (azimuth, range) float32 837kB ...
│           ...                 ...
│           DB_DBZE8           (azimuth, range) float32 837kB ...
│           sweep_mode         <U20 80B ...
│           sweep_number       int64 8B ...
│           prt_mode           <U7 28B ...
│           follow_mode        <U7 28B ...
│           sweep_fixed_angle  float64 8B ...
├── Group: /sweep_1
│       Dimensions:            (azimuth: 360, range: 581)
│       Coordinates:
│           elevation          (azimuth) float32 1kB ...
│           time               (azimuth) datetime64[ns] 3kB 2022-08-09T19:03:57.87900...
│           longitude          float64 8B ...
│           latitude           float64 8B ...
│           altitude           float64 8B ...
│         * azimuth            (azimuth) float32 1kB 0.03021 1.187 2.054 ... 358.0 359.1
│         * range              (range) float32 2kB 1e+03 1.3e+03 ... 1.747e+05 1.75e+05
│       Data variables: (12/17)
│           DBTH               (azimuth, range) float32 837kB ...
│           DBZH               (azimuth, range) float32 837kB ...
│           VRADH              (azimuth, range) float32 837kB ...
│           WRADH              (azimuth, range) float32 837kB ...
│           ZDR                (azimuth, range) float32 837kB ...
│           KDP                (azimuth, range) float32 837kB ...
│           ...                 ...
│           DB_DBZE8           (azimuth, range) float32 837kB ...
│           sweep_mode         <U20 80B ...
│           sweep_number       int64 8B ...
│           prt_mode           <U7 28B ...
│           follow_mode        <U7 28B ...
│           sweep_fixed_angle  float64 8B ...
├── Group: /radar_parameters
│       Dimensions:    ()
│       Coordinates:
│           longitude  float64 8B -71.33
│           latitude   float64 8B 4.564
│           altitude   float64 8B 206.0
├── Group: /georeferencing_correction
│       Dimensions:    ()
│       Coordinates:
│           longitude  float64 8B -71.33
│           latitude   float64 8B 4.564
│           altitude   float64 8B 206.0
└── Group: /radar_calibration
```

Similarly, I tried to perform a similar approach using netCDF files from the ERAD short course hosted in the Pythia s3 bucket.

```python
import xradar as xd
import fsspec

URL = "https://js2.jetstream-cloud.org:8001/"
path = f"pythia/radar/erad2024"

fs = fsspec.filesystem("s3", anon=True, client_kwargs=dict(endpoint_url=URL))
file = fs.glob(f"{path}/dda_data/*")[:1][0]
with fs.open(f"{file}", "rb") as f:
    dt = xd.io.open_cfradial1_datatree(f.read())

```

However, I got the following error:

```python 
Traceback (most recent call last):
  File "/snap/pycharm-community/425/plugins/python-ce/helpers/pydev/pydevd.py", line 1570, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/snap/pycharm-community/425/plugins/python-ce/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/media/alfonso/drive/Alfonso/python/xradar/delete.py", line 145, in <module>
    main()
  File "/media/alfonso/drive/Alfonso/python/xradar/delete.py", line 142, in main
    open_fsspec()
  File "/media/alfonso/drive/Alfonso/python/xradar/delete.py", line 53, in open_fsspec
    dt = xd.io.open_cfradial1_datatree(f.read())
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/media/alfonso/drive/Alfonso/python/xradar/xradar/io/backends/cfradial1.py", line 349, in open_cfradial1_datatree
    ds = open_dataset(filename_or_obj, engine=engine, **kwargs)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alfonso/mambaforge/envs/raw2zarr/lib/python3.12/site-packages/xarray/backends/api.py", line 671, in open_dataset
    backend_ds = backend.open_dataset(
                 ^^^^^^^^^^^^^^^^^^^^^
  File "/home/alfonso/mambaforge/envs/raw2zarr/lib/python3.12/site-packages/xarray/backends/netCDF4_.py", line 666, in open_dataset
    store = NetCDF4DataStore.open(
            ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alfonso/mambaforge/envs/raw2zarr/lib/python3.12/site-packages/xarray/backends/netCDF4_.py", line 421, in open
    raise ValueError(
ValueError: can only read bytes or file-like objects with engine='scipy' or 'h5netcdf'
```

I wonder why `open_iris_datatree` supports "data streaming" while `open_cfradial1_datatree` apparently doesn't.

kmuehlbauer, 2024-11-08:
open_cfradial1_datatree uses netcdf4 backend I think, and that doesn't support this?

aladinor, 2024-11-08:
Yeap. I think we might need to use `kerchunk` to make those files look like ARCO format. What about NEXRAD @kmuehlbauer?

closed 2024-11-08 by aladinor