openradar/xradar #247: TypeError: only integer scalar arrays can be converted to a scalar index

wreed1989, 2024-11-23:
I'm currently working on a project where I'm working with raw LiDAR data that comes in .hpl format and I'm struggling to get xradar to read in the file. 

I used the following code adapted from the example here: https://docs.openradarscience.org/projects/xradar/en/latest/notebooks/HaloPhotonics.html

Code:
`hpl = 'User1_100_20240714_122137.hpl'
rhi = xd.io.open_hpl_datatree(hpl, backend_kwargs = dict(latitude = 42.7452, longitude = -86.0961, altitude = 213))`

Data file format:

Filename: User1_100_20240714_122137.hpl
System ID: 100
Number of gates: 833
Range gate length (m): 12.0
Gate length (pts): 4
Pulses/ray: 3000
No. of waypoints in file: 1
Scan type: User file 1 - csm
Focus range: 65535
Start time: 20240714 12:21:45.80
Resolution (m/s): 0.0382
Range of measurement (center of gate) = (range gate + 0.5) * Gate length
Data line 1: Decimal time (hours) Azimuth (degrees) Elevation (degrees) Pitch (degrees) Roll (degrees)
f9.6,1x,f6.2,1x,f6.2
Data line 2: Range Gate Doppler (m/s) Intensity (SNR + 1) Beta (m-1 sr-1) Spectral Width
i3,1x,f6.4,1x,f8.6,1x,e12.6,1x,f6.4 - repeat for no. gates
**** Instrument spectral width = 0.305763
12.36262778 90.00 -0.00 -0.46 -1.77
0 0.1529 1.339292 1.910213E-5 0.0382
1 -0.6115 1.239853 1.350899E-5 0.0382
2 -3.0576 1.074272 4.185253E-6 0.0382
3 -14.1033 1.056070 3.161522E-6 0.0382
4 2.4079 1.034895 1.969022E-6 0.0382
5 2.7519 1.015216 8.592881E-7 16.3583
6 3.0959 1.038732 2.189486E-6 18.2311
7 1.4524 1.035328 1.999168E-6 29.4297
8 2.5990 1.022900 1.297460E-6 13.1478
9 6.8797 1.036261 2.057073E-6 20.6773
10 1.6053 1.027125 1.540954E-6 19.6453

However, I keep getting this error and I'm not sure how to address it.  Any guidance on how to get beyond this hurdle? I'd sure appreciate it. 

---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
File /opt/anaconda3/envs/devel/lib/python3.10/site-packages/xarray/backends/file_manager.py:211, in CachingFileManager._acquire_with_cache_info(self, needs_lock)
    210 try:
--> 211     file = self._cache[self._key]
    212 except KeyError:

File /opt/anaconda3/envs/devel/lib/python3.10/site-packages/xarray/backends/lru_cache.py:56, in LRUCache.__getitem__(self, key)
     55 with self._lock:
---> 56     value = self._cache[key]
     57     self._cache.move_to_end(key)

KeyError: [<class 'xradar.io.backends.hpl.HplFile'>, ('/Volumes/LaCie/University of IL/ATMS 590 Individual Study/MITTEN-CI/Obs/July 14/071424_IOP9/20240714/User1_100_20240714_122137.hpl',), 'r', (), 'e02fb1ca-fe8a-47a2-a1d1-dfc8aeed83c5']

During handling of the above exception, another exception occurred:

TypeError                                 Traceback (most recent call last)
Cell In[48], line 2
      1 hpl = '/Volumes/LaCie/University of IL/ATMS 590 Individual Study/MITTEN-CI/Obs/July 14/071424_IOP9/20240714/User1_100_20240714_122137.hpl'
----> 2 rhi = xd.io.open_hpl_datatree(hpl, backend_kwargs = dict(latitude = 42.7452, longitude = -86.0961, altitude = 213))

File /opt/anaconda3/envs/devel/lib/python3.10/site-packages/xradar/io/backends/hpl.py:638, in open_hpl_datatree(filename_or_obj, **kwargs)
    636         sweeps.extend(sweep)
    637 else:
--> 638     sweeps = _get_h5group_names(filename_or_obj)
    640 ls_ds: list[xr.Dataset] = [
    641     xr.open_dataset(filename_or_obj, group=swp, engine="hpl", **kwargs)
    642     for swp in sweeps
    643 ]
    645 dtree: dict = {
    646     "/": _get_required_root_dataset(ls_ds, optional=optional).rename(
    647         {"sweep_fixed_angle": "fixed_angle"}
   (...)
    653     "/radar_calibration": _get_radar_calibration(ls_ds, radar_calibration_subgroup),
    654 }

File /opt/anaconda3/envs/devel/lib/python3.10/site-packages/xradar/io/backends/hpl.py:584, in _get_h5group_names(filename_or_obj)
    583 def _get_h5group_names(filename_or_obj):
--> 584     store = HplStore.open(filename_or_obj)
    585     return [f"sweep_{i}" for i in store.root.data["sweep_number"]]

File /opt/anaconda3/envs/devel/lib/python3.10/site-packages/xradar/io/backends/hpl.py:440, in HplStore.open(cls, filename, mode, group, **kwargs)
    437 @classmethod
    438 def open(cls, filename, mode="r", group=None, **kwargs):
    439     manager = CachingFileManager(HplFile, filename, mode=mode, kwargs=kwargs)
--> 440     return cls(manager, group=group)

File /opt/anaconda3/envs/devel/lib/python3.10/site-packages/xradar/io/backends/hpl.py:435, in HplStore.__init__(self, manager, group)
    433 self._manager = manager
    434 self._group = group
--> 435 self._filename = self.filename

File /opt/anaconda3/envs/devel/lib/python3.10/site-packages/xradar/io/backends/hpl.py:444, in HplStore.filename(self)
    442 @property
    443 def filename(self):
--> 444     with self._manager.acquire_context(False) as root:
    445         return root.filename

File /opt/anaconda3/envs/devel/lib/python3.10/contextlib.py:135, in _GeneratorContextManager.__enter__(self)
    133 del self.args, self.kwds, self.func
    134 try:
--> 135     return next(self.gen)
    136 except StopIteration:
    137     raise RuntimeError("generator didn't yield") from None

File /opt/anaconda3/envs/devel/lib/python3.10/site-packages/xarray/backends/file_manager.py:199, in CachingFileManager.acquire_context(self, needs_lock)
    196 @contextlib.contextmanager
    197 def acquire_context(self, needs_lock=True):
    198     """Context manager for acquiring a file."""
--> 199     file, cached = self._acquire_with_cache_info(needs_lock)
    200     try:
    201         yield file

File /opt/anaconda3/envs/devel/lib/python3.10/site-packages/xarray/backends/file_manager.py:217, in CachingFileManager._acquire_with_cache_info(self, needs_lock)
    215     kwargs = kwargs.copy()
    216     kwargs["mode"] = self._mode
--> 217 file = self._opener(*self._args, **kwargs)
    218 if self._mode == "w":
    219     # ensure file doesn't get overridden when opened again
    220     self._mode = "a"

File /opt/anaconda3/envs/devel/lib/python3.10/site-packages/xradar/io/backends/hpl.py:326, in HplFile.__init__(self, filename, **kwargs)
    319 data_unsorted["azimuth"] = np.where(
    320     data_unsorted["azimuth"] < 360.0,
    321     data_unsorted["azimuth"],
    322     data_unsorted["azimuth"] - 360.0,
    323 )
    324 if data_unsorted["sweep_mode"] == "rhi":
    325     data_unsorted["fixed_angle"] = np.unique(
--> 326         data_unsorted["azimuth"].data[
    327             np.argwhere(np.abs(diff_azimuth) <= transition_threshold_azi) + 1
    328         ]
    329     )
    330 elif (
    331     data_unsorted["sweep_mode"] == "azimuth_surveillance"
    332     or data_unsorted["sweep_mode"] == "vertical_pointing"
    333 ):
    334     data_unsorted["fixed_angle"] = np.array(unique_elevations)

TypeError: only integer scalar arrays can be converted to a scalar index

syedhamidali, 2024-11-26:
@wreed1989 You can upload it to your google drive or onedrive box.com and share a link. Make sure the link is accessible and post it in this thread.

wreed1989, 2024-11-26:
Here are a few links to a few of each file type I'm trying to work with in order of importance - User1, Processed Wind, Stare.

User1 Files
https://drive.google.com/file/d/1oUJQM-49b2w16vlf3PLymM8TjArwju05/view?usp=drive_link
https://drive.google.com/file/d/18R9ZjFnSJo11VLrxYD3BEIV3rzZ_QifS/view?usp=drive_link
https://drive.google.com/file/d/1P2oVV8rpUUwk0xIMXxR2ppY4bLlvQiCG/view?usp=drive_link

Processed Wind Files
https://drive.google.com/file/d/13ShJE6pfFX0KToRdD_8espIXAhG6Ok3r/view?usp=drive_link
https://drive.google.com/file/d/1MkfAT7BA-TkoxNoOQGs6_n64Askt2e7i/view?usp=drive_link
https://drive.google.com/file/d/1NBK6T42L3B5R4uNY6cUkdUAxkZ5_xRUu/view?usp=drive_link

Stare Files
https://drive.google.com/file/d/1h8QkOS4Ktdv4eRN8evIc65xXMAFn7PDy/view?usp=drive_link
https://drive.google.com/file/d/1F28D9LfDUc3Z0NWgZ8CYITbz0ZSm8ak1/view?usp=drive_link
https://drive.google.com/file/d/1RWf14NoyqjKMXozJJMhjDk9tucptBT41/view?usp=drive_link

wreed1989, 2024-11-26:
Oops. Forgot to adjust the settings. 

Attempt 2:

User1 Files:
https://drive.google.com/file/d/1oUJQM-49b2w16vlf3PLymM8TjArwju05/view?usp=sharing
https://drive.google.com/drive/folders/1kfwO1h3iAouv1_HmyPrblV1RfxlaAE_P?usp=sharing
https://drive.google.com/file/d/1P2oVV8rpUUwk0xIMXxR2ppY4bLlvQiCG/view?usp=sharing

Processed Wind Files:
https://drive.google.com/file/d/13ShJE6pfFX0KToRdD_8espIXAhG6Ok3r/view?usp=sharing
https://drive.google.com/file/d/1MkfAT7BA-TkoxNoOQGs6_n64Askt2e7i/view?usp=sharing
https://drive.google.com/file/d/1NBK6T42L3B5R4uNY6cUkdUAxkZ5_xRUu/view?usp=sharing

Stare Files:
https://drive.google.com/file/d/1RWf14NoyqjKMXozJJMhjDk9tucptBT41/view?usp=sharing
https://drive.google.com/file/d/1F28D9LfDUc3Z0NWgZ8CYITbz0ZSm8ak1/view?usp=sharing
https://drive.google.com/file/d/1h8QkOS4Ktdv4eRN8evIc65xXMAFn7PDy/view?usp=sharing

rcjackson, 2024-11-26:
#250 fixes this issue with reading your RHIs (User1 files). The Stares should already work fine. For the processed wind files, I'd recommend just opening them with pandas [read_csv](https://pandas.pydata.org/docs/reference/api/pandas.read_csv.html). Skip the first line and specify a whitespace separator when doing so. The first column of the file is height, the second is wind direction in degrees, and the third is wind speed in m/s.

closed 2024-11-27 by mgrover1

syedhamidali, 2024-11-27:
@wreed1989 You can now update your xradar to include the latest changes, which allow you to read these HPL files. To update, activate your environment, and run:
```
python -m pip install git+https://github.com/openradar/xradar.git
```

PR #250 (2024-11-27): FIX: RHIs now work with Halo Photonics Reader
This fixes the error shown in PR #247 when an RHI was being loaded by the Halo Photonics reader.

- [x] Closes #247 
- [x] Changes are documented in `history.md`