openradar/xradar #259: NexradLevel2 Bug related to conflicting sizes for dimension

kmuehlbauer, 2025-01-30:
tracked also in #258 

### Bug related to conflicting sizes for dimension
- Previously unreported bug.
- This error does not occur in the `pyart` reader !
- In the MCVE below, we first decompress the file.
- Error: `ValueError: conflicting sizes for dimension 'azimuth': length 360 on 'azimuth' and length 358 on {'azimuth': 'DBZH', 'range': 'DBZH'}`

```python
import os
import fsspec
import xradar as xd
import subprocess

local_tmp_dir = "/tmp"
fs_args = {}
_ = fs_args.setdefault("anon", True)
fs = fsspec.filesystem("s3", **fs_args)

s3_filepath = "noaa-nexrad-level2/2010/01/01/KABR/KABR20100101_000618_V03.gz"
local_filepath = os.path.join(local_tmp_dir, os.path.basename(s3_filepath))
fs.download(s3_filepath, local_tmp_dir)
subprocess.run(["gzip", "-d", local_filepath], check=True)

dt = xd.io.open_nexradlevel2_datatree(local_filepath.replace(".gz", ""))
```

kmuehlbauer, 2025-01-30:
The reader breaks when trying to access the last sweep:

```python
import xarray as xr
ds = xr.open_dataset(local_filepath.replace(".gz", ""), engine="nexradlevel2", group="sweep_6") 
ds
```

All sweeps before that will read perfectly fine. Investigating the root cause.

mgrover1, 2025-01-30:
Ohh thanks for raising the issue here!

kmuehlbauer, 2025-02-03:
So, I've found the root cause of the ValueError:

https://github.com/openradar/xradar/blob/830d86b1c6290f1dce0e73c60a1d3b819735f906/xradar/io/backends/nexrad_level2.py#L1240-L1246

In the above code we retrieve the number of rays by calculating 

`nrays = n_end - n_start + 1 - n_intermed`

where `n_intermed` is the numbers of intermediate records within the data records. In the above case in the final record, those intermediate records are after `n_end` and should not be subtracted. So the solution is to count the number of intermediate records which fall inside `n_start`` and `n_end`.

PR #261 (2025-02-04): FIX: correct retrieval of intermediate records in nexrad level2 reader

- [x] Closes #259
- [ ] Tests added
- [x] Changes are documented in `history.md`
closed 2025-02-04 by kmuehlbauer