openradar/xradar #260: ODIM reader bugs when sweeps have different set of variables

ghiggi, 2025-01-30:
I noticed that the ODIM `open_odim_datatree` reader is not able to open files where the variables varies across the sweeps. 
This is a the common case for example for FMI radars `fivan` and `fiika`. Here below I provide a MCVE which might help to solve the issue. 

```
import os
import fsspec
import xradar as xd

local_tmp_dir = "/tmp"
fs_args = {}
_ = fs_args.setdefault("anon", True)
fs = fsspec.filesystem("s3", **fs_args)

s3_filepath = "s3://fmi-opendata-radar-volume-hdf5/2021/01/01/fivan/202101010055_fivan_PVOL.h5"
s3_filepath = "s3://fmi-opendata-radar-volume-hdf5/2021/01/01/fiika/202101010055_fiika_PVOL.h5"
local_filepath = os.path.join(local_tmp_dir, os.path.basename(s3_filepath))
fs.download(s3_filepath, local_tmp_dir)

# BUG
dt = xd.io.open_odim_datatree(local_filepath) # ValueError: These variables cannot be found in this dataset:  [...]
```

The following code show the variability across sweeps and across files.

```
local_tmp_dir = "/tmp"
fs_args = {}
_ = fs_args.setdefault("anon", True)
fs = fsspec.filesystem("s3", **fs_args)

filepaths = ['s3://fmi-opendata-radar-volume-hdf5/2021/01/01/fiika/202101010055_fiika_PVOL.h5',
 's3://fmi-opendata-radar-volume-hdf5/2021/01/01/fiika/202101010100_fiika_PVOL.h5',
 's3://fmi-opendata-radar-volume-hdf5/2021/01/01/fiika/202101010105_fiika_PVOL.h5',
 's3://fmi-opendata-radar-volume-hdf5/2021/01/01/fiika/202101010110_fiika_PVOL.h5',
 's3://fmi-opendata-radar-volume-hdf5/2021/01/01/fiika/202101010115_fiika_PVOL.h5',
 's3://fmi-opendata-radar-volume-hdf5/2021/01/01/fivan/202101010055_fivan_PVOL.h5',
 's3://fmi-opendata-radar-volume-hdf5/2021/01/01/fivan/202101010100_fivan_PVOL.h5',
 's3://fmi-opendata-radar-volume-hdf5/2021/01/01/fivan/202101010105_fivan_PVOL.h5',
 's3://fmi-opendata-radar-volume-hdf5/2021/01/01/fivan/202101010110_fivan_PVOL.h5'
]

for filepath in filepaths:
    local_filepath = os.path.join(local_tmp_dir, os.path.basename(filepath))
    fs.download(filepath, local_tmp_dir)
     
    
    # Open with xarray and determine number of sweeps
    dt = xr.open_datatree(local_filepath)
    n_sweepes = len([sweep_name for sweep_name in list(dt) if sweep_name.startswith("dataset")])
    
    # List variables in each sweep 
    for i in range(0, n_sweepes):
        sweep_name = f"sweep_{i}"
        ds = xr.open_dataset(local_filepath, group=sweep_name, engine="odim")
        print(sweep_name, ":", list(ds))
    print("-----")
```
------------------------

Finally, when asking for an unexisting sweep, xradar raise not a very useful error. 
For example, calling `xr.open_dataset(local_filepath, group="sweep_21", engine="odim")` raise `KeyError: 'dataset22'`. 
Maybe we can return a more informative message with list of available sweep acroynms `sweep_*` ?

kmuehlbauer, 2025-01-30:
@ghiggi Since late last year DataTree is first class citizen of xarray. We would need to implement `open_datatree` on our ODIM backend (and for all other backends). 

For the variable not found error, this is most likely a bad retrieval of needed root variables from the sweep variables.

ghiggi, 2025-01-30:
Yeah I had the same thought when I looked at the code base and how much it can be simplified using `open_datatree`. 

For the variable not found error, the error occurs in  `_get_required_root_dataset`. 
 
```
dt = xr.open_datatree(local_filepath)
n_sweeps = len([sweep_name for sweep_name in list(dt) if sweep_name.startswith("dataset")])

# List variables in each sweep 
ls_ds = []
for i in range(0, n_sweeps):
    sweep_name = f"sweep_{i}"
    ds = xr.open_dataset(local_filepath, group=sweep_name, engine="odim")
    print(sweep_name, ":", list(ds))
    ls_ds.append(ds)
 
_get_required_root_dataset(ls_ds, optional=optional)
```
 
Unfortunately I don't have the time right now to dig and work on fixing these bugs ... but I wanted to at least report the ones that I encountered.

kmuehlbauer, 2025-01-30:
No worries @ghiggi, your efforts are much appreciated.

egouden, 2025-02-03:
This bug has been fixed already ;-)
See https://github.com/openradar/xradar/pull/252.

kmuehlbauer, 2025-02-04:
@ghiggi Can you confirm, that this works on latest main? Then we can close. Thanks @egouden, for some reason I couldn't not connect this issue with your PR.

closed 2025-02-04 by ghiggi