openradar/xradar #272: `rainbow/backend may be using wrong string for `startrange` for `start_range`

Steve-Roderick, 2025-02-11:
As an example, consider the RAINBOW file used in the notebook: https://github.com/openradar/xradar/blob/main/examples/notebooks/Rainbow.ipynb

```
filename = DATASETS.fetch("2013051000000600dBZ.vol")
ds = xr.open_dataset(filename, group="sweep_0", engine="rainbow")
```

Inspecting the file `2013051000000600dBZ.vol` carefully:

```
$ grep -a startrange 2013051000000600dBZ.vol
$ # No match
```

Notice the following search for (`start_range`) instead.

```
$ grep -a start_range 2013051000000600dBZ.vol
         <start_range>0</start_range>
         <start_range>0</start_range>
```

Consider the source https://github.com/openradar/xradar/blob/main/xradar/io/backends/rainbow.py#L720
```
        start_range = self.root._get_rbdict_value(
            var, "startrange", default=0, dtype=float
        )
```

I believe the start range may not always be populated correctly and some rainbow files may not decode at all.

The rainbow spec may prescribe `start_range` and not `startrange`.

kmuehlbauer, 2025-02-11:
@Steve-Roderick Thanks for the well-written issue. I agree, it looks like this isn't handled properly. This is probably from my impression that `stoprange` corresponds with `startrange` and not with `start_range`. When implementing this in wradlib back in the days, there have been no field for `start_range`. 

@Steve-Roderick Would you be up to fixing this?

Steve-Roderick, 2025-02-14:
Thank you @kmuehlbauer for confirming this issue. I am up for fixing this issue - thanks for suggesting this. Beyond correcting the xml tag name to `start_range`, I will attempt to verify that the `start_range` is correctly applied. I believe this would be the distance in km to the center of the zeroth range bin (this `start_range > 0` implies a cone of silence on the sweep with the lowest elevation angle.

kmuehlbauer, 2025-02-14:
> I believe this would be the distance in km to the center of the zeroth range bin (this `start_range > 0` implies a cone of silence on the sweep with the lowest elevation angle.

Currently, we assume the reference plane for `start_range` and `stoprange` to be the beginning of the range bin. Indeed, for `start_range` > 0 this would imply a cone of silence. There is no mention in my documentation what the reference plane actually is, but we should go with the most logical approach. Please note that for xradar we use bin-center as reference.

syedhamidali, 2026-10-02:
@Steve-Roderick are you still planning to work on this? If not, I would be happy to take it over for the v1.0 milestone: read `start_range` (falling back to `startrange` for older files) and treat it, like @kmuehlbauer described, as the start of the first bin, so that the internal `range` stays bin-center. No worries either way, just let me know.

closed 2026-10-07 by kmuehlbauer