openradar/xradar #275: Vaisala IRIS Sigmet Header 8708

CocoasColas, 2025-05-10:
Hi all,

I've been trying different solutions to read this Sigmet file, and it seems no matter what I do whether it be PyArt or Wradlib, I cannot read this file for the life of me. I will provide a link that contains the Sigmet files, I have no clue if you guys can come up with a solution to read this.

https://drive.google.com/drive/folders/1LY27PaQ7WiHDwnw27k0go5WyiHO2MN63?usp=sharing

aladinor, 2025-05-11:
Hi @CocoasColas. Thanks for reporting this. Could you post here the code you are using for opening your SIGMET files and the error you are getting?  This will give as an idea of what might be wrong.

CocoasColas, 2025-05-11:
Hi @aladinor.

So, I was utilizing the standard reader from the Documentation on Wradlib for the XRadar backend for the IRIS support, so at the moment the script looks like this, I added the header reading to check out what was being read directly and both he sigmet reader and the bytes returned 8708. When I read it with PyArt, it told me that it did not support negative dimensions:

`import wradlib as wrl
import matplotlib.pyplot as plt
import numpy as np
import warnings
warnings.filterwarnings("ignore")

fpath = "../OUN040530003040.RAWRH5H"
with open(fpath, "rb") as f:
    header = f.read(4)
    print(int.from_bytes(header, byteorder="little"))
fcontent = wrl.io.read_iris(fpath)

# which keywords can be used to access the content?
print(fcontent.keys())
# print the entire content including values of data and
# metadata of the first sweep
# (numpy arrays will not be entirely printed)
print(fcontent["data"][1].keys())
print()
print(fcontent["data"][1]["ingest_data_hdrs"].keys())
print(fcontent["data"][1]["ingest_data_hdrs"]["DBZ"])   
print()
print(fcontent["data"][1]["sweep_data"].keys())
print(fcontent["data"][1]["sweep_data"]["DBZ"])`

kmuehlbauer, 2025-05-12:
@CocoasColas After some imvestigation this is no current sigmet/IRIS data file. So it can't be decoded with the current iris backend.

Looking at the first bytes it seems similar to UF format (universal format). The header seems almost identical (beside the "PF", instead of "UF"). Unfortunately, even after changing all "PF" -> "UF", also PyART isn't able to digest this file (`pyart.io.read_uf(fpath)`).

@CocoasColas It would be great if you could provide a bit more information on the provenience of this file. 

I think we should now definitely work on ingesting UF format. If someone is brave enough to try...

CocoasColas, 2025-05-12:
Hi all,

The source of this data comes from the NSSL Thredds server, as it hosts the KOUN archive. The data is labeled as RAW data, yet the data file naming format is very similar to those of IRIS SIGMET. This is the URL containing the data;(https://data.nssl.noaa.gov/thredds/catalog/RRDD/KOUN/2004/RAW/OUN20040530/catalog.html). I had assumed it was SIGMET because NSSL has a history of utilizing their software, such as the NOXP Mobile Radar utilizes Vaisala SIGMET processing. Looking closer into the Hex data, it does mention IRIS at the start of what I would assume is each sweep with what looks to be a version number 7.29, and then products possibly? I haven't a clue what to do with this information, or if it'll help at all. 

[image]

kmuehlbauer, 2025-05-12:
Thanks for attaching that little image of the first bytes. 

I've tinkered with this for a while today, and I'm sure this is a variety of Universal Format (UF) which is also implemented in the IRIS software suite. I've already checked with the IRIS Programmers Manual, and indeed, the file contains 360 rays of data with 7 moments (also seen in the lower part of the provided image (CZ, VR, SW, DR ...). But what concerns me is the "PF" instead of "UF". If someone can come up with an explanation for that, would be great.

I've started a test implementation for an UF reader and if my time allows, I might be able to have a prototype ready in a few days.

CocoasColas, 2025-05-12:
Hi,

Thanks so much for taking the time to consider this, a lot of this data is considered "lost media" if you will, the WDSS-II files we have are incredibly low resolution and aren't really reliable for any research or future projects, so this is a big help. I haven't a clue what the PF instead of UF could be, I don't recall any statements or knowledge of them utilizing a custom proprietary version of any radar software to produce these files.

I look forward to seeing if any results can be produced in the future.

PR #276 (2025-07-11): ENH: add universal format (UF) backend

- [x] Closes #275  
- [x] Tests added
- [x] Changes are documented in `history.md`
- [x] notebook and docs added 

This adds a first UF backend implementation. No fancy stuff for now. This should work already:

```python
import xarray as xr

fname = "NAMMA_NPOL_VOL_QC_060930_0000.uf"
ds = xr.open_dataset(fname, group="sweep_0", engine="uf")
print(ds)
```
```
<xarray.Dataset> Size: 14MB
Dimensions:            (azimuth: 257, range: 751)
Coordinates:
    elevation          (azimuth) float64 2kB ...
  * range              (range) float32 3kB 100.0 300.0 ...  [... cut]
closed 2025-07-11 by kmuehlbauer