openradar/xradar #279: xradar citation

ghiggi, 2025-06-25:
Hello ! 

I am citing xradar in a manuscript and I am not sure how to cite it correctly.
Do we want to specify an "How to cite" section somewhere in the documentation or README? 
Currently I cite xradar with: 

`Mühlbauer, K., M. Grover, A. Ladino, E. Goudenhoofdt, Z. Sherman, H. A. Syed, and R. Jackson, 2025: xradar - A tool to work with weather radar data in xarray, https://doi.org/10.5281/zenodo.7091736.`

Additionally, I saw that the last archived version of the software on Zenodo is v.0.0.2 (of March 24, 2023) ... with many authors/contributors which are missing. 

@kmuehlbauer @mgrover1 @aladinor @syedhamidali @egouden @rcjackson

kmuehlbauer, 2025-06-25:
I mean the zenodo link is directly on the startpage of the README.md and with that also in the first page of the docs.

We could add a bullet point above the about-section:

- Please cite xradar: zenodo-link

kmuehlbauer, 2025-07-09:
@ghiggi I've added a section to the docs which is hopefully comprehensive enough. See #284

PR #284 (2025-07-11): MNT: package and docs

- [x] Closes #273
- [x] Closes #282  
- [x] Closes #279
- [x] Changes are documented in `history.md`

This removes some unneeded files from the package and fixes the docs in that regard.

- flake8, tox, Makefile, MANIFEST.in 
- add explicit note on cartesian data
- add How to cite section on README/docs
closed 2025-07-11 by kmuehlbauer